jstevensog/wixel-sdk #3: how was the maximum_battery and min batt calculated

mayaalinder, 2015-03-31:
Hello.. I have a question on these two lines:
#define BATTERY_MAXIMUM	(1771)
#define BATTERY_MINIMUM	(1416)
Is it 1771 mVolt? and how did you calculate it?

jstevensog, 2015-04-02:
No, this is the value for each limit where 0-2048 is 0-100% of the
calibration voltage of the ADC.  To calculate the value for a particular
voltage input to the voltage divider, you need to use a calculation similar
to this.
value =
(((VoltageIn/VoltageDividerTotalResistance)*VoltageDividerMinimumResistance)/1.25v)*2048

Where:
VoltageIn = the amount of voltage input to the Voltage Divider. that you
want to calculate a value for.
VoltageDividerTotalResistance = 37000 ohm
VoltageDividerMinimumResistance = 10000 ohm
1.27V = The calibration reference of the ADC.
2048 = the maximum value of the ADC at the ADC Voltage Reference

Hope this helps.

I am working on code that will store these limits in Flash memory, so a
full charge/discharge cycle of the LiPo will set these values for you.

Cheers

mayaalinder, 2015-04-07:
So in your calculation, did you use 4.0V as the VoltageIn for Battery_Maximum? and 1.25v or 1.27 as calibration ref?
I build my system with a 2.2k ad a1.0k resistor in the voltage divider.. the battery says max charge is 4.2V and min 2.75V

jstevensog, 2015-04-07:
I used 3.9V as maximum, & 3.2V as minimum. Reason being that a lipo fully charged to 4.2V will rapidly discharge to around 3.9V, the slowly discharge to around 3.5V, then more rapidly to cut off. I have not yet found a lipo that discharges to the rated 2.75V. My one discharges to 3.2V, which is the rated standard voltage of a lipo, so I used this as an "average" lipo minimum. 
Voltage reference is the wixel internal voltage reference, which I don't remember off the top of my head, and am not in front of my computer right now. It is in the wixel doc. 
Cheers

 
John Stevens
M - +61 418526349

jstevensog, 2015-04-08:
Internal voltage reference is 1.25V.  You can use VDD, but it is less
reliable.  I therefore chose to use the internal 1.25V reference, and the
voltage divider I documented.
Cheers

mayaalinder, 2015-04-08:
:-) I calculated backwords from 1771.. 1771/2048*1.25/10*37=3.999
I wish  bought a couple of extra batteries.. we have a very nice battery test rig at work..

jstevensog, 2015-04-09:
Great.  If you gain a better understanding of how to determine the lipo
battery capacity, let me know.  I would like to improve this so that it is
easy for everyone that uses dexbridge2.
I may have mentioned, I am planning to make the bridge code handle a full
battery cycle and set the upper and lower limits in flash.  That way it
should be easy for anyone to do a full charge/discharge cycle and have a
relatively accurate capacity indication.
Cheers

closed 2015-07-28 by jstevensog