jstevensog/wixel-sdk #4: Type of resistor for xBridge2

ceben80, 2015-06-20:
Hi John,

regarding to this link

https://github.com/StephenBlackWasAlreadyTaken/xDrip/blob/gh-pages/hardware_setup.md

1/4w 1K Ohm and a 1/4w 2.2K Ohm resistor should be used.

But you mentioned in the xBridge2 PDF 27k and 10k resistors...

What type are you prefer for the xBridge2 hardware setup?

Thanks, Matthias

jstevensog, 2015-06-20:
Hi Mathias,
I prefer the larger resistors I specified. They mean less drain on the battery. You can use the original values, but you will need to alter the source code and recompile. 
Cheers

 
John Stevens
M - +61 418526349

ceben80, 2015-06-20:
Great!

 

One Question regarding the BT module…any concerns regards with the HM-11?

And, Adrian mentioned, if the HM-11 will work the baud rate must adapt to 115000, right, also in the xBridge2 Code?

 

 

Regards,

Matthias

jstevensog, 2015-06-20:
Well, the module is not the issue, it is the firmware on the module. Some are delivered with a firmware or post QA set at a different baud rate. You will need to set it to 9600 to make it work with this wixel code. Next version will be adding auto detect of baud rate and will store this in flash. A lot more coding to do. The instructions about putting the latest firmware on are in the document (PDF). Setting the baud rate is easy to do by issuing the correct commands. If you have an issue I will try to help. I have only used the HM-10 modules so far. 
Cheers

 
John Stevens
M - +61 418526349

ceben80, 2015-06-20:
Thanks John!

...now waiting for HM-11 parcel delivery. Once all components are being soldered together I'll give you a feedback.

Cheers
Matthias

jstevensog, 2015-06-20:
Your welcome Mathias. Hope it gets there soon. 
Cheers

 
John Stevens
M - +61 418526349

ceben80, 2015-06-20:
One question to the values to be altered in the source code when using the 27k and 10k resistors.
...didn't find the the exact code position, sorry for that :-)

Thx, Matthias

jstevensog, 2015-06-21:
Hi Matthias,
Sorry for the delay in responding. 
There are two lines near the top of the source file that define values for BATTERY_MAXIMUM and BATTERY_MINIMUM. You comment out these two lines, and uncomment the two lines below. 
Cheers

 
John Stevens
M - +61 418526349

ceben80, 2015-06-21:
np John, thanks for your answer - found in the code

...I will close this issue

Cheers
Matthias

closed 2015-06-21 by ceben80