jstevensog/wixel-sdk #5: hm-11 firmware flash impossible

LadyViktoria, 2015-07-15:
hi
i tried to update the hm-11 to the latest firmware. but when i type AT-SBLUP i did not get OK-SBLUP. All other AT commands are working.

jstevensog, 2015-07-15:
Hi @LadyViktoria,
I have not tried to flash a HM-11, I have always used HM-10 modules from
JNHuaMao Technology Company in China.
Their flash upgrade documentation states explicitly that the AT+SBLUP
command is used for HM-10/11/15 modules.
I have heard of others that have had this issue with HM-11 modules.  I
believe that whomever they bought the modules off has supplied a module
with firmware that is not genuine.  A problem with all the different
Chinese suppliers.

If the module functions, I would advise not bothering to upgrade.  The most
important command for the module to understand is the AT+NAME command.  The
latest xBridge code will automatically detect the Baud rate of the module
on first boot up, and use that exclusively from then on.

If you wish to persist with upgrading the firmware, I suggest you contact
the supplier and ask them to provide the correct instructions to put the
module into boot loader upgrade mode.  When you find out, please pass them
on to me so I can update the document to include them if the standard
instructions don't work.  I will also contact JNHuaMao and ask them about
the issue with the HM-11 modules, in case they know a fix for older/test
firmware.

Cheers

LadyViktoria, 2015-07-15:
hi
i have this one
http://www.seeedstudio.com/depot/Bluetooth-V40-HM11-BLE-Module-p-1803.html

jstevensog, 2015-07-15:
Hi @LadyViktotia,
I can only say it looks like a hm-11, but I cannot say that it is from the original manufacturer.  I have asked JNHuaMao about this, but I believe these may be copies. I would send an email to seedstudio asking them how to update the firmware. Also, can you send me the AT+NAME? & AT+VER output please?
Thanks in advance. 

---- LadyViktoria wrote ----

LadyViktoria, 2015-07-15:
ok
AT+NAME? -> AnyFlite
AT+VERs? -> v528

jstevensog, 2015-07-15:
Thanks,
This is not a genuine hm-11, and that is why the firmware upgrade procedure will not work. If I were you, I would give it a go as is.  xBridge should be able to work with it fine. Let me know how you get on. Also, get seedstudio to either supply you with a genuine module, or provide the instructions to update the firmware. 
Good luck. 
Cheers

---- LadyViktoria wrote ----

jstevensog, 2015-07-16:
Hi @LadyViktoria,
I just found this site.
http://www.seeedstudio.com/wiki/Bluetooth_V4.0_HM-11_BLE_Module.  It shows
all the AT commands this version of the module supports.  Unfortunately,
there is no boot loader or firmware update commands.  Only a very small
subset of what is available on the genuine HM-11 module.

If it functions OK for you, keep using it, and maybe order a new one either
direct from
https://www.fasttech.com/product/1740900-hm-11-bluetooth-v4-0-transceiver-ble-module
just in case.  I have found these guys reliable in the past.  They have
only ever sent me genuine HM-10 modules, so I expect the HM-11 from them
would also be genuine.  They have also been very good to replace a module I
thought was faulty (even though it was fine, and they really needn't have
done so).
Sorry for the problems.
Cheers

closed 2015-07-28 by jstevensog