jstevensog/wixel-sdk #6: xBridge2 (lastest code from 07/20/2015) sudden signal gaps

ceben80, 2015-07-21:
Hi John,

I have been soldering new componts for xBridge2 (accoring the xBridge2.pdf) using the lastest code provided yesterday. Unfortunately I have sudden signal gaps.

I have very good soldering experiences, all solder joints look very good to me.

I have been running a xDrip Experimental box (different hardware) since 6 weeks without any gaps.

[image]

My setup consists of: 
HM-10 (version 540) from 

http://www.ebay.de/itm/New-HM-10-CC2540-4-0-BLE-Bluetooth-to-UART-Transceiver-Module-Central-Peripheral-/390879063376?pt=LH_DefaultDomain_77&hash=item5b02352d50

Sony xperia E1 Android 4.4.4

Pololu Wixel

https://www.pololu.com/product/1336

Here you can see all componets temporarily placed in this open box due to testing purpose. Always the green box is very close to the Dex sensor/transmitter.

[image]

I would be very happy to have to stability of "dexdrip.c" which is used for xDrip Experimental.

Thanks for your assistance John!

Greeting from Germany,
Cheers

Matthias

JoernL, 2015-07-21:
Yes, same issue for me, too. There were several lags and the battery service life was the same as with Wixel-BT connection. LG L40 with Android 4.4.4 + Sunfounder HM10 (V540). I've tested it some weeks ago ...

Thanks for your work and commitment.

Greets

jstevensog, 2015-07-21:
Hi Matthias,
That is strange.  If you go back one commit, does stability improve?  My current production bridge has very occasional packet drops, sometimes more often than others.  I'll send a link when I get to work.  I may not get to look into it for a couple of days.
Cheers

---- ceben80 wrote ----

ceben80, 2015-07-21:
Thanks John for investigation. If I can anything do for you, please let me know!

One thing regarding the battery capacity...I am using a 250 mAh Lipo, but I feel that the battery is going to be empty in less than 20 hours...same as JoernL mentioned above.

btw, from next sunday on I'll be on vacantion for 2 weeks without having access to the internet...Sometimes not bad to relax :-)

Cheers,
Matthias

jstevensog, 2015-07-21:
Hi Matthias,
I use 1200mAh lipo. I get around 6 days between charges. Long battery life is more important to me than small size. 
Can you tell me if you are using the latest xDrip-Experimental? A lot of packet drop issues were caused by that code. Also, if you can plug into a computer and connect using a terminal program, you will be able to capture some logs if what is going on. That may help me figure out where the issue is. 
Cheers

---- ceben80 wrote ----

jstevensog, 2015-07-21:
Hi Matthias,
Here is my nightscout site for you to take a look at.

https://cloudcgm.azurewebsites.net/

As you can see, there are occasional packet drops, but overall it is very
reliable.
I have looked into the differences between the xBridge2 and xBridge3 (where
I am doing development) code and there is nothing there that would impact
packet capture.
I will look into it further.
Cheers

ceben80, 2015-07-21:
I am using the xDrip E version #90 where you have been merged Adrian's statistic module.
Okay I will enable "Store logs forf troubleshooting"...same location on my smartphone as the sqllite file were stored?

Cheers
Matthias

ceben80, 2015-07-21:
One question regard to the Android version if this could be a BT related issue...I don't...
Does this work better with Lollipop? I heard Lollipop has a more reliable BT stack?!

jstevensog, 2015-07-21:
Actually, if you can capture the wixel logs, that would be better.  You
need to connect the wixel to a computer using USB, and it will appear as a
serial (COM) port.  You can use the wixel configuration utility to figure
out which port it attaches as.
Then run say PuTTy or Hyperterm or any other serial terminal program,
connect to the COM port the wixel is using,
You should also have an option in the program to store a session log into a
file.  You will need to sit near the computer for an hour or so, or carry
the laptop and keep it near you, until you see a gap in xDrip-E.

You can press the S key to get a status at any time.  This will show you
the version of code, and a few other things.
Cheers

ceben80, 2015-07-21:
perfect! Will do this tomorrow and come back to you John.
Thanks for your time, Cheers Matthias

jstevensog, 2015-07-21:
Yes, lollipop (Android 5+) is much more stable with BT.  xBridge works like
this.
1) Wakes up. Powers on HM-1x module
2) Listens for a Dexcom Packet.  Does not sleep until it gets one.
3) Once it gets a packet, waits for the BLE to be established.  If BLE is
not established 2 minutes since the packet was captured, gives up and goes
to sleep.
4) Once BLE is established, send the data.  Wait for an ACK packet back
from xDrip.  If it doesn't get an ACK back in 2 minutes since the packet
was captured, gives up and goes to sleep.
5) Once it gets the ACK, it saves settings to flash (if it needs to),
calculates it's sleep time, and goes to sleep.

If BLE does not get established in 2 minutes from packet capture, it will
appear to be a packet drop in xDrip.
If xDrip gets confused and thinks it has already processed a packet (VERY
rare/never happens now we have wakelocks in the code) it will appear to be
a packet drop in xDrip.

BTW, if you want better reliability, wire the HM-1x to the wixel like the
classic xDrip, and use the resistors specified there as well.  You will
lose a some battery life, but the code will think that it is running on an
xDrip circuit and NOT depower the HM-1x, so BLE will always be established
unless you go out of range.  The wixel will still sleep between packets,
giving you some power savings.
Cheers

jstevensog, 2015-07-21:
You are welcome.
I'm on my 1 hour train trip to work.  I would be bored otherwise ;)
Cheers

ceben80, 2015-07-21:
Ok, in this setup I don't have the chance to change the wires due to fixing the wires with glue.
But regarding my first classic xDrip I only need to solder the resistors...when I can try to use the xBridge2.wxl

ceben80, 2015-07-22:
Hello John!

Here is the trial in terms of putty logging...looks a little bit strange regarding the output characters.
The dex signal was coming...I see a new BG value on the apps, but the putty session is going into a freeze state...no feedback.
[image]

Cheers,
Matthias

jstevensog, 2015-07-22:
Hi Matthias,
Whenever the wixel goes to sleep (sleeping for 260), the putty session will
freeze.  However, if you simply keep it open, it will resume when it wakes.
Yes, the strange characters are there because it sends the binary data
packets to and from xDrip out the serial port as well as vie BLE.  This was
useful in developing the protocol, but I should probably remove it at some
point.
If you can grab say a 30 minute period, with a couple of packet drops, it
will show me where the problem lies.
Thanks for doing this.
Cheers

ceben80, 2015-07-22:
Wow..thanks for the speedy response - great
btw- what's you local time now?

Will make another trail...come back to you again,

Cheers,
Matthias

jstevensog, 2015-07-22:
It's around 15:45 here, so mid afternoon/evening.  Just got back in the
office and fired up the laptop and saw your mail.
Cheers

ceben80, 2015-07-22:
Grrr...Signal Missing 21 min. ago

[image]

Putty Log:

https://drive.google.com/file/d/0B8T1Je5277XYU3R4TndHZGFVV2M/view?usp=sharing

ceben80, 2015-07-22:
Absolutly strange...after 31 min Missing signal I decided:

- disconnect wixel from PC
- turn off/on the lipo

...now BG value is coming...I have no idea

[image]

ceben80, 2015-07-22:
Hi John,

to be honest, I'm desperate with this setup...

I can see four or five values on my phone, but putty doesn't any logging information?

But now again "Missed Signal 40 Minutes ago"...

Okay, will disconnect Wixel from the PC again and finally press "Forget Device" and Scan BT again...nothing.

Disconnect the lipo and connect again...

Now the signals are coming again, totally crazy.

Sorry would be happy to give you more useful input regarding this issue :-(

Yesterday it was still running more  reasonably

Cheers,
Matthias

[image]

ceben80, 2015-07-22:
I then changed back to classic xDrip. As you can see on this screenshot, no signal dropouts.
I would be very happy if you see any possiblity to fix this issue...would be great John. If I can anything test for you, please let me know.

Cheers,
Matthias

[image]

AdrianLxM, 2015-07-22:
I'm experiencing the same problems with the classic setup running xBridge.
So I guess it's not the BLE that causes the package drops, but the wixel.

(Switched back to xDrip for holidays as I needed a reliable system and now came back with a cold - will take some time until I can switch back for debug logs)

As Jörn, Matthias and I built the xDrip at around the same time and are all from Austria/Germany, chances are high, we are using the same lot of wixels.

ceben80, 2015-07-22:
Hi Adrian,

Thanks for joining in this issue. I firmly believe that we can work together to solve this problem, everything will be fine, the concept of xBridg2 is so fantastic.I'd love to use it reliably.

Cheers,
Matthias

jstevensog, 2015-07-22:
Hi All,
I have just ordered a new wixel so I can build a classic xDrip-wixel
circuit.  I already have an HM-1x module, so once it arrives and I have
some time, I will be able to start work looking into what is going on.  My
gut feeling is that either there is some issue with the Sleep time
calculation, or in the logic determining when to send the packet on to
xDrip.
I am looking into to code in the meantime, but nothing is standing out.
Cheers

ceben80, 2015-07-23:
Thanks John for further investigation!

Best Regards,
Matthias

jstevensog, 2015-07-26:
Matthias, and everyone else having this issue, can you please try this
wixel file on your bridges?
https://github.com/jstevensog/wixel-sdk/blob/master/apps/xBridge2/xBridge2-test.wxl
I am hoping it fixes things for you.  I am running it on my production
bridge, and performance is the same, if not slightly better.

Cheers

NoahT72, 2015-07-26:
thanks John,

Will try it tonight.

Noah

ceben80, 2015-07-27:
Hello John,

due to vacation I only have my xDrip classic at this moment...
I will do further tests from August 10th on...I will give you feedback once I have finished the testing stuff.

Thanks again for investigation John!!!

I'll get back to my vacation

Cheers,
Matthias

JoernL, 2015-08-06:
Hi John,

i've tried out the latest xbridge2 app and there is something strange.
The yellow led comes up every 10 minutes. Sometimes after 7 minutes.
Maybe there is something wrong with my HM-11 module. I've soldered
the STATE cable to the LED connector. Is this correct? When the yellow led
comes up, does it mean - i am listening? ... or ... i got some packets?

Cheers

JoernL, 2015-08-06:
Update: I've tested it the last 2 hours and mostly every 10 minutes i got a reading -
sometimes it takes 15 minutes. But never 5 minutes.

jstevensog, 2015-08-06:
Yellow LED indicates the BLE connection is up between HM-1x and phone.  This should occur about every 5 minutes.  When it goes off, this means that a packet was captured and sent to the phone, and the wixel has gone to sleep.  Are you using a classic circuit or xBridge circuit?
Cheers

---- JoernL wrote ----

jstevensog, 2015-08-06:
That is very strange.  I am thinking that these wixels use a slower Crystal clock speed, which would upset the sleep cycle.  That or there is something wrong on the phone side.  Kit Kat and earlier do not consistently reconnect.  Can you try connecting a terminal program to the wixel via USB and capturing the output for me?
Cheers

---- JoernL wrote ----

JoernL, 2015-08-07:
I am using xBridge circuit. Can you please tell me your way of monitoring the output?
Putty always freeze on COM3 after go to sleep for 260sek. Do you connect the battery,
too or only USB? Yesterday i have tried a little bit playing with some values in the code.
When i connect it to the battery, yellow led comes up and get txid. Then the first packet
was catched fine ... go to sleep ... yellow(off). The second time it wakes up, fetches the
package ... tx over BT yellow(on) ... and sleeps again ... yellow(off). All good!
But from this moment on the wixel doesn't wake up after 260sek or fetches something.
I had to wait ... 10min ... sometimes OK ... sometimes 15min.
If the clock speed is really different, please tell me on which lines can i do a "try & error"
of some values in the code.

Cheers

AdrianLxM, 2015-08-07:
If the clock speed of the wixel was different, wouldn't this also affect the xDrip-wixel programme that runs perfectly fine on those wixels?

jstevensog, 2015-08-07:
I usually run battery connected after the wixel is started by USB connection, so that the battery capacity is reported.

 When the wixel sleeps, there will be no response to any command on the terminal, and there will be no output.  When sleeping, the wixel is essentially torned off.

Once you have disconnected the USB, always power cycle the wixel by disconnecting and reconnecting the battery.  When connected to USB, it uses a different sleeping mode to allow the USB to remain connected.  It is not compatible with the normal sleep mode.

Can you please capture the output from a wixel with unmodified code in a log file and send it to me?

I do not understand why it does not wake sometimes.  Please do not alter values in the code. They are very precise, and changing them may hide the true problem.  Please capture output from a wixel that is running the unchanged code, and preferably newly flashed.

Adrian is sending me a wixel of his so I can try for myself.

---- JoernL wrote ----

jstevensog, 2015-08-07:
Also, it really did not make sense that the first two packets are captured fine, meaning that the first dleepnis successful, and then subsequent sleep cycles are not consistent.  Mine is always consistent.  Also not, that the 260 seconds is only an indication of how long it will sleep.  
The wixel calculates actual sleep by calculating the time since the packet was captured and sleep initiated, and subtracting that from 260 seconds.
That way, it will always wake in time.

I will make another bridge tomorrow to test further myself against the last committed code.  In case some error has crept into it.  I am sure I am running the same code on my production wixel.
Cheers

---- JoernL wrote ----

jstevensog, 2015-08-07:
Perhaps, but I am not sure if the xDrip wixel code sleeps in the same way.  It is strange that waking and packet capture are incinsistent.  Joern is using the xBridge circuit, and having the same problem. Yet others, including myself are not.  It will either be an inconsistent Crystal oscilator on the wixel, or a code issue.  Difficult to diagnose remotely.
Cheers

---- AdrianLxM wrote ----

JoernL, 2015-08-07:
Yes, i can capture the output, but it has no errors ;) The capture from putty only exists till
the first sleep. After that none output on COM3 - even the wixel wakes up and fetches the 
second packet. So please tell me your way of capturing.

jstevensog, 2015-08-07:
Joern, that is my wah of capturing.  My wixels have no problem waking up and writing to PuTTY.  The fact that yours does makes me suspect there is something wrong with the wixel.  I always capture the debug messages this way, and have done so using several wixels with no problem.

---- JoernL wrote ----

JoernL, 2015-08-07:
Yep, that's the problem. My capture freezes after the first sleep time.

jstevensog, 2015-08-07:
It sounds like the wixel is going into the wrong sleep mode.  Sleep mode 1 keeps the USB alive, but sleep mode 2 does not.  I need to sleep, it is near midnight here.  I will look into how the wrong sleep mode can be selected tomorrow, and confirm all my PuTTY settings as well, and let you know.

---- JoernL wrote ----

JoernL, 2015-08-08:
I've played a little bit and figured out that when i comment out the line

//IRCON &= 0x7F;

in function ISR(ST, 1)

Then the wixel wakes up much better. After ignnition mostly every 5 minutes.
Later the sleep time decreases. Sometimes the wixel go to sleep and after
60sec wakes up and yellow led turns on till it fetches the next packet and
again go to sleep.

jstevensog, 2015-08-09:
H Joern,
Sorry for taking so long, the weekend was spent resolving some family
matters.
I am now on the train to work, so I have a bit of time.

My PuTTY settings are nothing special.  115200, 8 bits, 1 stop bit, no
Parity, and XON/XOFF flow control (default), but I also use no flow
control..  The wixel app looks for the ACM_CONTROL_LINE_DTR to determine if
the USB is connected, and if it is, it uses PM1 to sleep.  Otherwise it
uses PM2, which is the normal operation.

PM1 allows the USB to stay connected during sleep, it is a slightly more
"awake" state.  PM2 kills the USB connection and breaks output to the
terminal.

It is dangerous to remove the line that sets IRCON &=0x7F from the ISR
function.  This line is important as it clears the Sleep Timer interrupt
flag (the first thing the ISR must do) without affecting any other
interrupt flag.  If this is NOT done, the ISR will be retriggered and may
continue to be retriggered indefinitely.  The results of this are difficult
to predict.

Reading the CC2511 documentation, this would likely be better to be IRCON
&=0x3F; to ensure that bit 6 of IRCON is set to 0, and I will make this
adjustment to the code and test once more.  If bit 6 of IRCON is ever true,
it will cause the ISR to continuously retrigger without ever exiting, so it
is a good thing I found this out.

If you wish to alter your code, please set
IRCON&=0x3F;
in the ISR(ST,1) function.  Let me know the result.

However, I am still unsure why your wixel is killing the USB port when it
sleeps.  This is VERY strange behaviour, and seems to indicate it is going
into the wrong sleep mode.
Cheers

JoernL, 2015-08-09:
Hi John,

okay 0x3F. I will try that later tomorrow, cause first i have to replace my transmitter
battery. 

Cheers

jstevensog, 2015-08-10:
All,
Can I please ask you to confirm if you are using a Mac, Linux, or Windows
to put the app on your wixels?
I am using a PC, and I automatically convert the .wxl file for use with
Mac.  I am wondering if perhaps something is going wrong with that
conversion and corrupting the program.
Also, are you building your own with my wixel-sdk?  Or are you using the
pre-made xBridge2.wxl file?  If you can do both, is there any difference
between building your own and using the pre-made one?

I am trying to find differences between what I use to create and put the
app on the wixel, and what you do.
There shouldn't be any issue, but I simply cannot explain why it is working
for me, and not for you.
Thanks in advance.
Cheers

jstevensog, 2015-08-10:
Also, I have pushed up a version of the code that includes the IRCON mask
change from 0x7F to 0x3F as discussed with Joern.
I doubt it will improve what you are seeing, but feel free to give it a go.
Cheers

AdrianLxM, 2015-08-10:
I use Ubuntu 12.4 with your precompiled .wxl.

I also experienced the same problems when building a modified xDrip (initialization and the 2 methds ported) built with the xDrip-Environment.

zjedr, 2015-08-10:
Precompiled on windows.
I'm not getting as many missed readings as others, but when I've tried running "restart collector" from within xDrip, it seems to get the following reading. (only done a few times so could be coincidence)

ceben80, 2015-08-10:
Hi John,

back from vacation...
I've always been using "xBridge2.wxl" - my Environment is running on W7.
Otherwise, can I try something to help you?

Is it true that there are differences between the European Wixel we are using (Adrian, Joern and myself) and the US version? That would be really strange.

Best wishes for today,

Matthias

jstevensog, 2015-08-10:
I do not think so, but there is obviously some difference.  When I make the
xBridge2.wxl file, I also process it so that it will load on Linux/Unix and
Mac as well as windows.  I use Windows normally, as my laptop is windows.
I also use a Ubuntu Virtual Machine to build the Pebble watch face, so also
have the Wixel SDK on that.  I have successfully loaded and run the xBridge
app from that environment in the past, but not recently.  I currently have
my Ubuntu virtual machine configured with the Standard wixel SDK, so I can
work on making the xBridge2.c code able to be compiled with a the standard
wixel SDK.

Either there is an issue with the wixels you guys have received, or there
is something else going on.  I am not sure yet, and Adrian is being very
generous in sending me one of your wixels to work on a solution.

Until I can reproduce what you guys are seeing, I can't figure out what is
going on.  Sorry for the long and protracted solution.

The new latest push of code with the IRCON mask being set to 0x3F is
working as well as the previous one on my production bridge.  Starting
manufacture of a spare bridge to take away with me to South America.  I
will let you know how that goes.
Cheers

JoernL, 2015-08-10:
0x3F doesn't solve the problem. After ignition it fetches the packet and goes to sleep.
Sometimes it wakes up after 5 min but sometimes not.
Disabling the line - it wakes up every time, but randomly. 30sec, 2min, 3.5min ...
I've compiled it from wixel-sdk with windows 7.
With gentoo linux and sdcc-3.5.0 it doesn't compile, because:

libraries/src/usb_cdc_acm/usb_cdc_acm.c:355: error 98: conflict with previous definition of 'usbComRxReceive' for attribute 'type'
from type 'void function ( const-unsigned-char xdata* fixed, unsigned-char fixed) fixed'
  to type 'void function ( unsigned-char xdata* fixed, unsigned-char fixed) fixed'
Makefile:172: recipe for target 'libraries/src/usb_cdc_acm/usb_cdc_acm.rel' failed

jstevensog, 2015-08-10:
Hi Joern,
I didn't expect it to solve the issue.  It is simply a safer mask.  With this line commented out, I expect variability, because the interrupt request control register (IRCON) is not being correctly reset.

Also, right now, you MUST use all of my version of the wixel-sdk on my github repo.  It contains changes to the standard libraries, and a couple of new libraries, that need to be compiled and linked in.

I am very concerned that your wixel is not keeping the USB alive during sleep, and I believe there may be an issue with it.  I don't have a spare to send you right now, and am waiting for Adrian's one to arrive so I can investigate that.

Sorry this is taking so long.
Cheers

---- JoernL wrote ----

seattlebrighton, 2015-08-11:
I'm building the xbridge version now, to try to help troubleshoot, but I'm using an hm-11. Should the hm-11 LED signal go into Status (P1-2 on wixel), since there's no output marked Status?

ceben80, 2015-08-11:
Yes...regarding to the diagram below the status LED is connected to P1_2

[image]

JoernL, 2015-08-11:
John, which values have to be changed that the wixel deep-sleeps in PM1? (Only for testing)

jstevensog, 2015-08-11:
Hi Joern,
If you wish to ensure that only PM1 is entered for sleep (USB connection
remains, although it will not respond while the wixel sleeps), I would line
1764 from:
usb_connected = state & ACM_CONTROL_LINE_DTR;
to:
usb_connected = 1;

This will make the wixel always believe that the USB is connected, and will
therefore enter PM1 on sleep.

You will not be able to send commands to the wixel while it is sleeping,
but it should maintain the USB connection during sleep, and begin writing
to the terminal once it wakes up.

Cheers

seattlebrighton, 2015-08-13:
Excellent. It's working with xbridge2 and xdrip-experimental beta 2.0.1.
Can I see the battery % somewhere?

jstevensog, 2015-08-13:
Bridge battery % should appear on the xDrip main screen, just below the delta figure.  It should be green until it is below 50%, yellow below 30% (IIRC, its been s while since I looked at that code).
Cheers

---- seattlebrighton wrote ----

seattlebrighton, 2015-08-15:
Below the delta figure, there's a red "waiting for packet" text. That's all
I see, and I'm using this week's beta experimental with xbridge2

jstevensog, 2015-08-15:
OK.  That means that the wixel is not seeing the battery voltage, or the battery voltage it is seeing is too low.  The value coming to xDrip is 0.  Check that you have soldered the resistors correctly.  The analog inputs of the wixel can only handle a maximum 3.3 volts. The resistors form a voltage divider to drop the battery voltage down into this range.
You will need to measure the voltage between the Gnd and P0_0 pins and make sure it is lower than 3.3 volts, and higher than 2 volts.
Cheers

---- seattlebrighton wrote ----

seattlebrighton, 2015-08-16:
Just measured from ground pin on wixel to P0_0 and got 2.997 VDC.
Everything else is working great. Ah, i see an update. Thanks. I'd like to
help Ben with Omnipod, too, unless you know of someone else who's further
along with Omnipod...

[35 more replies]

closed 2016-07-01 by jstevensog