jstevensog/wixel-sdk #9: Feature request - bridgestatus in mongo similar to devicestatus

jweismann, 2015-10-28:
Hello John

I would like to analyze the bridge2 battery discharging in more details and
thought it would be convenient to have the bridge battery uploaded to mongodb
just like we have the uploader phone battery in the devicestatus collection
today. Would it be possible to add another collection to mongo, say bridgestatus,
that would have battery data for the bridge instead of the device.

Thanks a lot in advance,

Cheers, Jacob

jstevensog, 2015-10-28:
HI Jacob,
I like the idea, and have had it in the back of my mind for some time.  I
will see what can be done, but as it is part of xDrip, not the xBridge2
code, it may not make it in until the next xDrip release.
One thing to keep in mind though, is that xBridge2 battery capacity is a
rough indication only.  The reason for this is that it is determined from
the battery voltage delivered to the wixel.  The problem with this is that
it is not a linear function, nor is it an accurate determination of battery
capacity.
I am looking at alternative capacity determination and circuits based on
actual current input/output from the battery, but this will also be
difficult with the wixel sleeping, and again may not be accurate.
My preferred option would be a smart charger board that did all of that,
and could be communicated via some sort of serial bus.  We still have ports
on the wixel to do this.  We could then simply interrogate the charger
board for the capacity when the wixel wakes, or before it sends data to the
phone.  This would be far more accurate, and the charger board would likely
have thermal protection of the battery, which is sadly missing from the one
we have been using thus far.

Anyway, I will keep it in mind when I am next looking at the bridge code,
and see if it can be easily added.
Cheers

jweismann, 2015-10-30:
Hi John.

Great and thanks again! 

As to the changes to xdrip, I am using the published beta version (latest apk from https://github.com/StephenBlackWasAlreadyTaken/xDrip/wiki/xDrip-Beta) and was trying to locate the source related to this beta version in order to look into this. I guess that it's based on a tag from https://github.com/StephenBlackWasAlreadyTaken/xDrip-Experimental but I'm not quite sure. I would be grateful if you could provide me with the proper link to the source code used for the xDrip-beta releases.

As to the capacity - your points above explains my interest in the data :) One of my aims was to use the data to determine the discharge functions for the batteries that I use and once I have these discharge functions I can improve the function that relates battery voltage with the remaining capacity.  

Cheers, Jacob

closed 2016-07-01 by jstevensog