jstevensog/wixel-sdk #10: Diagram for bare HM11 seems inconsistent with the datesheet

jweismann, 2015-11-06:
Hi John

After the success with the HM10 module, I had a look at the bare HM11 diagram and the GND line from the HM11 is drawn as going out from line13 in the diagram in xbridge2.pdf but according to the datasheet, GND is line12 (line 13 is PIO3), cf. reference below. Have I misunderstood something or is there are typo in one of the documents ?

Cheers, Jacob

PS! Datasheet referece (section 1.7)
http://www.pridopia.co.uk/pi-doc/BT4.0-HM-10 Serial_Port_BLE_Module_Master_Slave.pdf

jweismann, 2015-11-06:
Hi John 

I can now confirm that soldering according to the the datasheet specification, i.e. using line12, works, so maybe there is a typo or several variants with different layout.

Cheers, Jacob

jstevensog, 2015-11-08:
Hi Jacob,
You are correct, so it appears it is a typo.  I will rectify it as soon as
possible.
Thanks and Cheers

closed 2016-07-01 by jstevensog