jstevensog/wixel-sdk #12: Delay Calculations

ClearSimpleAndPlain, 2016-01-04:
In the xBridge2.c file, line 1724 you have this code:
if(delay)
{
	delay - (channel*500);
}

Is it supposed to be "delay -= (channel*500)"?  (added the "-=" instead of "-")
Should this be taken out since it looks like it's been running without it anyway? Since no change to delay is happening?

Thanks,
Denis.

jstevensog, 2016-01-04:
I'll need to take a good look at it Denis.  The 500 is the ms gap in which a packet is expected in each channel, so my gut feeling is that the code is correct.  It has been a good 6 months since I looked at that section of the code in depth.

Essentially, the delay needs to have 500 * channel (0-3) subtracted from it to accurately calculate the sleep time.
Cheers

---- ClearSimpleAndPlain wrote ----

jstevensog, 2016-01-04:
Hi Denis,
It should be -= as you suggest.  I'll make the change and push it up.
Cheers

---- ClearSimpleAndPlain wrote ----

jstevensog, 2016-01-05:
Hi Denis,
Thanks for picking this typo up.  Not sure when it got in there, but I have
had way less dropped packets on my production bridge since I put it on this
morning.
Cheers

zjedr, 2016-01-06:
Yes, I've been running new version for about 24 hours and am also getting way fewer dropped reading.
Might be worth some notification to others xBridge2 users.
Good pick up.

ClearSimpleAndPlain, 2016-01-06:
Good to know. I'm just trying to get my head around the code. You say you still get dropped packets. Do you know the percentage?

zjedr, 2016-01-06:
I've missed 11 readings in last 24 hours (11 of 288), a couple of those might be valid. All misses are singles.

jstevensog, 2016-01-06:
I haven't dropped a packet (unless for a totally known reason) since I
upgraded my prod bridge this morning.  I do get dropped packets on
occastion, usually for one of two reasons:
1) Something interfering between the Dex transmitter and xBridge.  ie, too
many coins in my pocket AND my car keys, my wife's leg covering the
transmitter or bridge, I left the bridge in my other pants......
2) Something interfering between the bridge and the phone.  ie, wife's leg
over the bridge and phone elsewhere, xDrip not accepting the packet (rare,
but happens and I have to restart Bluetooth).

I had noticed that packets after a one received on a channel >0 were more
likely missed, and now I know why.  Likely the sleep time was too long, so
it missed the channel 0 packet and then had some chance of missing the
others.  With Denis' pickup of my typo, this won't happen.
BTW, I have posted in CGM in the Cloud FB page, and NS Australia FB pages.
Feel free to share to your local NS group pages.
Cheers

ClearSimpleAndPlain, 2016-01-06:
I was thinking about point #2 of your missing packets. You get an acknowledge after successfully sending the data from the bridge to the phone Via Bluetooth. If for some reason the bridge doesn't get the ack writhing some time, is it possible to save the data to the memory and send it the next time we have a good Bluetooth connection?

jstevensog, 2016-01-06:
Certainly possible, but the available ram is an issue, and xDrip needs to be able to handle a record with a ms_since_capture field.  Likely to happen in xBridge3, but not in xBridge2.
Cheers

---- ClearSimpleAndPlain wrote ----

JohanDegraeve, 2016-01-06:
@jstevensog is this link having the new versin ? https://raw.githubusercontent.com/jstevensog/wixel-sdk/master/apps/xBridge2/xBridge2.wxl

jstevensog, 2016-01-06:
It is as of about 11:00 AEST (00:00 GMT).  Cheers

---- Johan Degraeve wrote ----

closed 2016-06-30 by jstevensog