jstevensog/wixel-sdk #14: Shorter battery life by using a voltage divider (27k/10k - xDrip classic/xBridge2 firmware)

ceben80, 2016-01-19:
I would like to know the influence of the voltage divider with regard to the battery life.

I not really noticed a shorter battery life with my 1100mAh Lipo (it runs for 3.5 days), but I have been experiencing with a 250mAh Lipo a battery life reduction of 40 percent when using the 27k/10k voltage divider.

I am running xDrip classic soldering with the xBridge2.wxl

Can anyone tell similar issues or behavior?

I have been modified these values in xBridge2.c 

#define BATTERY_MAXIMUM_CLASSIC (1814)
#define BATTERY_MINIMUM_CLASSIC (1416)

Any help or advice will be appreciated!

Cheers,
Matthias

jweismann, 2016-01-19:
Hello Mathias

I had the same thoughts when I looked at the xbridge circuit. The voltage divider
constantly draws current in the circuit albeit it isn't much. To the best of our
knowledge there is no reason not to make this sleep as well between the readings
as the rest is doing (xbridge2 circuit, that is) and I discussed this with a HW
engineer that came up with the proposal attached. This proposal does require a
simple code change too, i.e. one with have to set the white pin high when it the
wixel awakes. I have soldered the prototype for the proposal above but have not
had time to test it.

The solution I use for "production" is an xbridge2 with a 500mAh Lipo from Adafruit
and with xbridge2 soldering. The battery drops to around 50% (but not linear drop
and from 50% it drops very fast) in approximately 2 days.

A 40% reduction for your 250mAh solution due to the voltage divider doesn't
sound right to me.

Is there a particular reason that you don't run with the xbridge2 circuit? The
sleeping between readings should prolong your battery lifetime.

Cheers, Jacob
[image]

jstevensog, 2016-01-19:
Hi Matthias,
The voltage divider is required because the analogue input of the wixel
will only handle a maximum 1.2V input.  Higher than that will damage the
wixel input.  Lipo are generally 3.5-3.8V.
Can you please measure the voltage between P0_0 and GND when the 250mAh
battery is fully charged and let me know what it is?

When you say a "reduction of 40%" battery life, what exactly do you mean?
Is it the number of days of run time?
Are you aware that the 250mAh Lipo will only last slightly less than 25% of
the 1100mAh Lipo?

The wixel draws 30mA when awake and operating (LEDs lit are additional to
this), and less than 1mA when sleeping.  The HM-1x module draws 50mA when
powered on.  The voltage divider draws a constant 0.001 mA at all times.
If one wixel LED is lit, estimate 10 mA of draw.
In an hour, with minimum current draw, the xBridge circuit will
theoretically be 5mA.  With an 1100mAh battery, this will be theoretically
9 days of operational life, or with a 250mAh battery, this will be
theoretically 2 days operational life.  So the expected difference between
the batteries running on an xBridge circuit would be a 78% reduction of
battery life for the 250mAh compared to the 1100mAh.

I hope this clears things up.  The Voltage Divider does draw a constant
current which has a minor impact on battery life.  The xBridge circuit and
program is designed to maximise battery life, and sleeping and powering off
the HM-1x module accomplishes the vast majority of this.

I do not know the design specifications of the xDrip-wixel circuit, but
assuming that there is no Voltage Divider, it should draw a theoretical
23mA.  This would mean an approximate 2 days of life with an 1100mAh
battery, and less than half a day with a 250mAh battery.

Cheers

ceben80, 2016-01-24:
Hi John,

sorry for my late response here:

to be clear, I am using not a xBridge soldering solution.
I am running with xDrip classic due to some Wixel chip issues.

Without the 27k/10k voltage divider the battery last approximatly 17 hours. 
By using the voltage divider the battery is empty after 10 hours.

When the battery is full charged the voltage between P0_0 and GND is 1,13 volt.

Cheers,
Matthias

jstevensog, 2016-01-24:
Hi Matthias,
If you are running xBridge2 on a classic xDrip circuit, you should use the original voltage divider resistors, not the 27k/10k ones. If xBridge2 has detected a classic circuit, it configured itself for the original values.  
I cannot remember the original resistor values, but there is quit a difference in the voltage divider ratio.
This would explain the differences you are seeing.
The 27k/10k resistors only draw around 89 micro amps, which would not cause an actual 70% reduction in battery life.

I believe the issue is entirely that xBridge2 is expecting values based on the classic circuit resistors, and you have used the xBridge2 circuit resistors.
This difference is likely the cause of the issue.  

I will try to do the calculations when I am off this bus to confirm this.
Cheers

---- ceben80 wrote ----

ceben80, 2016-01-25:
Thanks John for speedy reply as always!
I will try with 1k/2.2k voltage divider first with my huge box (1100er Lipo) due to the fact, this one is in use at the moment. 
I wonder if I can increase the duration of the battery life now.

Cheers,
Matthias

jstevensog, 2016-01-25:
It will depend on if "battery life" is the battery capacity indication, or actual time until the battery is flat.  I suspect it is the indication you are referring to? 
Otherwise, it will make it worse as those values draw more current.
Good luck.
Cheers

---- ceben80 wrote ----

ceben80, 2016-01-25:
I will keep an update once the 1100er Lipo is empty

ceben80, 2016-01-26:
Hi John,

a very strange behavior....
I am running with xDrip-classic and xBrigde2.wxl (original) NOW with the 1k/2.2k voltage divider.
For far so good, but for the last 24 h my battery display always shows 100%

Beautifully I have a battery that never goes empty :smile: 

[image]

Any idea?

Cheers,
Matthias

jstevensog, 2016-01-26:
Hi Matthias,
That one is very strange.  If you had been charging it, then the first time
after the wixel is loaded with xBridge2, I would expect it to show 100%
until fully charged, as it is learning your battery voltage limits.
But if you are not charging, then there is apparently something quite wrong.
Can you send me a photo of your bridge with these resistors?  I just want
to see if something is around the wrong way.  Then I will go through the
code and see if I have made any silly mistakes (happens often enough).
Thanks and Cheers

ceben80, 2016-01-26:
[image]

jstevensog, 2016-01-27:
Hi Matthias,
Ok, that didn't help me as much as I thought. ;)  The resistors are hidden
from view.  I add them slightly differently.  Here is an old photo of mine
using the 27k/10k resistors.

Just make sure that you have the lower value connected between GND and
P0_0, and the higher value between VIN and P0_0.
Confirm that for me, and during my lunch time I will look into the code to
see if I can spot an issue.
Cheers

ceben80, 2016-01-27:
As describted by Stephen the 1k goes to VCC, 2.2k to GND......that is my soldering solution.

Will mean this Doc is wrong......

ceben80, 2016-01-27:
Wow....it works. I just swap the wires and now have 91%

[image]

jstevensog, 2016-01-27:
Yep, I guess the document is wrong.
As wixel-xDrip hasn'e really used this before, I guess it was never picked
up.
I have only used xBridge so, I never did.  Just assumed that it was right.
Or, I just didn't check how it was wired, assumed the voltage divider was
much the same as mine, and calculated the wrong values.  I'll check to see
if it can work the other way, and if so, I change the code to suit, so
others don't have the same issue.
Cheers

ceben80, 2016-01-27:
Ok, I will see how the 1100er Lipo will last with the new voltage divider.
I will keep an update once the battery is empty. 

Thanks John so far.

Cheers,
Matthias

AdrianLxM, 2016-01-27:
Just to be clear that I understand it right:

@ceben80 used the 10k/27k version with a selfcompiled xBridge2 firmware where he defined

```
#define BATTERY_MAXIMUM_CLASSIC (1814)
#define BATTERY_MINIMUM_CLASSIC (1416)
```
(xBridge max and min for the classic setup)
... but left the rest the same.

And it still has a much faster battery drain?

ceben80, 2016-01-28:
In about 3 hours the 1100 mAh battery has a duration of 3 days behind and now shows 80%.
If the course still runs so linear, I think that the battery life is greater than 4 days (with 27k/10k)
...wow in this screenshot, the same bg as above :smile: 
[image]

Let's wait and see. I'll keep you posted

jstevensog, 2016-01-28:
It's definitely not linear.  While it will steadily decline, it flattens out from about 85-90% until 35%, then drops rapidly.  I use a 1200mAh lipo, and easily get 5-6 days.  Also have a 1000mAh Lion, that does about the same.
Cheers

---- ceben80 wrote ----

ceben80, 2016-01-28:
5 to 6 days would be great as compared to 3.5 days

fezulin, 2016-02-06:
Hi, I have a question quite near to the topic, you discussed here:
I used to run a xDrip-soldering with a pretty small battery (230mAh). That used to run for 17 to 19 hours. (With the xBridge2.wxl from January 2016.)
Then I changed the soldering: I now have a xBridge soldering with 10k/27k, still with the original xBridge2.wxl. 
Now my 230-mAh-battery runs for about 38 hours! (What is great!!)
I am just surprized, that the battery level starts with 63% - not 100%. Within the first 34 hours, the battery level declined from 63% to 30% in a pretty linear manner. Then, it took just 2 hours to 0%. 
The chractzeristic of declining is no problem at all. 
To start with something closer to 100% - would I have to change settings and use a self-compiled version of xBridge2.wxl (as AdrianLxM mentioned above)?
Do not misunderstand: I am happy, that my battery now runs THAT LONG! There is no real necessity to change anything. It is just a question of "high level comfort". :)

jstevensog, 2016-02-07:
OK, that is very strange. Is your battery a 3.7V 230mAh one?  The only reason for this is if one of the resistor values is incorrect.  The voltage divider ratio is important as it drops the voltage on the analogue input pin down proportionally.  
Generally, the adafruit charger boards charge to 4.2V, and xBridge2 stores this value (-1%) as the 100% value for the calculation. In fact, when first loaded, it is set to 3.8V, and as the battery is fully charged, it continues to save the maximum value it has "seen" less 1%. 
It does something similar with a falling battery voltage to "learn" the maximum and minimum charge voltages of your battery and charger board.

I would check the values of your resistors with a multimeter to be sure they are correct. It is very easy to make a mistake with the colour coding.
Hope this helps. 
Cheers

---- fezulin wrote ----

fezulin, 2016-02-07:
Thank you for your answer! You wrote "when first loaded". Eventually this might lead us to the reason of my "problem". I used other resistors before I ended up with my current version. First I had no resistors at all, then 1k/2.2k and now 10k/27k. I also made at least one mistakes in my soldering: HM10 on 3V3 instead of P1_0.
Is it really the values of "**first**" loading that are relevant for future displayed values? As I have been using different resistors this "first time loading" was likely done with wrong resistors. What would I have to do, to have a "first time loadfing" again? 
To be honest - I did not even took away the battery, when I changed the resistors. So the Wixel was uninterruped connected to power supply. 
I will disconnect the Wixel from the Battery for some moments and reconnet it. So I hope, if the battery is fully charged "first time" after the Wixel restarted, the values might be closer to 100%, if I recharge it the second time.

jstevensog, 2016-02-07:
"First loaded" means when you configure the wixel with xBridge2.  So every time you load xbridge2 on the wixel, the battery limits are set back to the defaults, and they are then "learnt" again through a charge/discharge cycle. If you have not loaded xbridge2 since you changed the resistors, then this is likely the problem.
Load xbridge2 onto the wixel again, and do a full charge, and it should correctly set the 100% value. A full discharge will set the 0% value.
Cheers

---- fezulin wrote ----

fezulin, 2016-02-07:
That sounds as if we are straight on the way to solve the problem. :)
Battery level was displayed with 46%. I just copied the xBridge2.wxl on the Wixel. Battery Level is now displayed with 71%. As the value changed, we are obviously doing something right. :)
Now I am going to start charging the battery.
Just to be sure: It will take a full cylce, to learn the values, correct? So I after the battery is charged, the device has to run, until the battery is completely empty. Correct? Then after charging next time the values, displayed in the xDrip-App should be correct.
As the battery will run for about 38 hours, it will take so long until I could tell you, what is displayed.

jstevensog, 2016-02-07:
Well, technically a full cycle is required.  However, the low limit will
not be "learnt" below a hard coded level to ensure that the wixel does NOT
write to flash (where it stores the learnt limits) if there is a chance
that it will lose power in the process and corrupt the memory.
More important is the full charge.

I have found that my battery depletes rapidly from 30% onwards, so I do
ensure I charge the bridge before this level.
Cheers

fezulin, 2016-02-10:
Hi,
what happened: I did load the original xBridge2.wxl on my Wixel.
I charged the battery. I let the xDrip run until the battery was completely empty.
Then I charged the battery again.
While charging the battery, the battery level was displayed with 100%.
After I disconnected the device from the charger, the battery level changed from 100% to 83%.

Well, I would consider this to be fine. There is no necessity to start with 100%. And 83 is at least much better then something about 60%.
It is sufficient for me, if I have an indicator about battery level. 
I will take care to charge the battery, if the battery level is about 30%. From 30% it took just 1 hour to 0%.
But with xBridge2.wxl my small 230 mAh battery was running again for more then 30 hours.

From my point of view: This issue can be considered to be solved.
Thank you for your help!

jstevensog, 2016-02-10:
Ah yes. I have said a number of times that measuring battery voltage is a VERY inaccurate way of determining it's capacity.  This is an example of why.

In order to charge the battery, the charger must apply a higher voltage to it.  So xbridge2 will display 100% well before the battery is fully charged.  It takes mine at least an hour after this to fully charge, and the charger indicates this with a green LED.

If I stop charging before this, depending on how close to fully charged it is, dropping to around 80% is not unusual.
In fact, even when fully charged, it will rapidly drop to around 90% in the first few hours.

It is an indication, not an accurate measurement, but it is better than nothing.

Good you are seeing the benefits of the battery saving.  Yesterday, I pushed a version that includes the "B" terminal command. This toggles on and off the HM-1x "depowering" while the wixel is sleeping.  If you are having issues with your phone's Bluetooth every bite and then, you can use this to turn off depowering the HM-1x, so it stays connected to the phone while the wixel sleeps.  The HM-1x looks like it is depowered (well, mind does) as the status LED goes out, but it's BLE radio status on.  This will impact battery life.  I am not sure by how much.  But you will get more reliable operation if your phone/Android version has trouble with the constant disconnection/reconnection.

Cheers

---- fezulin wrote ----

closed 2016-06-30 by jstevensog