jstevensog/wixel-sdk #16: xBridge2.wxl invalid data in file error

fredthesecond, 2016-01-25:
Have constructed the xBridge hardware with the potential divider and bluetooth state line and am trying to use the latest version of the xBridge2 software. Unfortunately I get an error in the Wixel configuration utility when using the xBridge2.wxl file within the apps/xBridge2 folder as follows:

Invalid data in file.
App File could not be opened: xBridge2.wxl

I've probably made a mistake somewhere along the line but I know my config utility is ok because I reprogrammed an xdrip a couple of days ago without a problem.

Any idea what could be happening?

Thanks,
Martin

jstevensog, 2016-01-25:
Hi Martin,
That is indeed strange. Are you using a Mac, or windows PC, or Linux?
It sounds as simple as a corrupt file. Originally, if those using Mac or Linux based PC tried to load the file, the standard "crlf" line terminators would upset them and give this error.  Are you compiling yourself, it using the wxl file from my repo?

My SDK make file runs the wxl file through sed and fixes this automatically (I use a windows laptop for this, which is how I found the error).

I would recommend downloading the xBridge2.wxl file from me repo, or downloading the whole wixel SDK in a zip file and extracting it to make sure it isn't corrupt.

Cheers

---- fredthesecond wrote ----

fredthesecond, 2016-01-26:
Hi

Sorry for the delay in replying.

I am using a mac and I think I've found the problem.  As you you seem to be saying - I was essentially 'Saving As' the xBridge2.wxl file and then using the resulting file. This process seems to be modifying the file during the operation making it incomprehensible to the config utility.

Having now downloaded the whole set of files using the proper download link, the same file is accepted by the config utility.

Thanks for your quick response,
Martin

jstevensog, 2016-01-26:
You're welcome Martin,
Enjoy.
Cheers

jstevensog, 2016-01-26:
BTW, I will also update the document to mention this for linux/mac users.
Cheers

xitation, 2016-01-27:
You both may find these utility's handy: https://en.wikipedia.org/wiki/Unix2dos you will need to source a copy for whatever flavour *nix distros your run including MacOS.

jstevensog, 2016-01-27:
Actually, I already use a post build pass through sed (gnuwin32 on my
windows box) to take care of it.  The files read fine on my linux VM as
well as Macs, but I think the issue was that a "save as" on the git repo
xBridge2.wxl causes the text to be reformatted and so the Mac/Linux wixel
config utility won't read it.  Not sure if running it through a dos2unix or
unix2dos utility would fix it, but it might.
I only use the windows build environment as I need to mostly use tools on
that.  Hence the Linux VM.  But if I am running the Linux VM I do a
pull/push from the repo and work there.  It's just a little more fiddly for
my morning commute. ;)
Cheers

closed 2016-06-30 by jstevensog

bordeb, 2018-01-22:
Hi! First, Jstevensog thank you on your great work!
I have the same issue today with latest 2.47e trying to write it to wixel.
When I try to open the .wxl app the Pololu Wixel Config Utility says exactly the same: 
_Invalid data in file.
App File could not be opened: xBridge2.wxl_
I am using RPi Raspbian to do that.
I tried to download through cli  using git , tried to download as zip file and than extract it on RPi, tried to do the same on Win and then transfer the file to RPi using Samba etc... Always the same error when I try to open wxl in the Wixel Utility.
Is there any solution for this?