jstevensog/wixel-sdk #20: No power when connected to lipo

thecubical, 2016-05-19:
I've wired it all up as per the pdf, however I haven't got the voltage resistors.

BAT and GND connect directly to VIN and GND on the wixel, for some reason the wixel doesn't appear to be powering on at all when the code is loaded, if i plug the wixel  in via usb it powers up correctly and works as its supposed to.

I've tried multiple batteries,wixels and lipo boards, is there something I'm missing? Even with usb power plugged directly into the lipo board it doesnt work.

If I erase the wixel and just try to run it via battery the LEDs light up, so at a guess it would seem like the app is modifying how the power gets to the board?

Any help would be great.

Cheers

jstevensog, 2016-05-19:
Hi Daniel,
The code deliberately does not use LEDs to display to much status. 30 second after you supply power, the red LED will flash. The LED on the HM-1x module will start to flash shortly after this.
The only other indications it is working, without connecting to a USB port of a computer and running a terminal program are as follows:
1) when you search for a bridge in xDrip, you see a device with a name like "xBridgeXXXX".  If you see a name like HMSoft, then there is a problem.
2) when you pair xDrip and your bridge, and the BLE connection is established for more than 1/2 second, the yellow LED will light and stay on.

This is by design to maximise battery life.  Never expect a wixel to behave in its default manner of blinking LEDs once you load an app on it.  That is entirely up to the app.

I am in vacation until next week.  So I will not be available to provide more detailed advice until then. Let me know how you get on.  I strongly recommend using the resistors.  If you are using this for OpenAPS, then I suggest you use xBridgeOaps instead.  It is also an n me repo.
Cheers

---- Daniel Powell wrote ----

thecubical, 2016-05-19:
It ended up working not quite sure what the issue was!

Thanks for the quick response.

closed 2016-06-30 by jstevensog