jstevensog/wixel-sdk #22: Power requirements - sleeping?

savek-cc, 2016-06-01:
I just measured the current actually drawn by the xBridge2 setup without voltage divider.
After boot while waiting for the first packet, it draws an almost constant 35mA.
Once a packet has been detected and sent to the phone, the orange LED is disabled (so no LEDs are on at this point) and BLE is disconnected (according to phone).
In this state, it still draws 10mA from the battery with current code (xBridge v2.42).
Status:
xBridge v2.42
dex_tx_id: 0123456 (XXXX)
initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 0, do_leds: 1
Any idea what might be going on while the wixel should be sleeping?

jstevensog, 2016-06-01:
Are you connected to USB? That is a lower sleep mode than when operating without a USB connection.  That current draw sounds like the lower sleep mode to me.

---- Timm Korte wrote ----

savek-cc, 2016-06-01:
USB was disconnected during the measurements.
Only attached battery + amp meter for the measurements.

jstevensog, 2016-06-01:
OK, that is strange.  Haven't actually measured my own bridges, but with the battery life I get it is fairly close to theoretical, so I haven't bothered.  By the time you count the occasional missed packets and staying awake to the next one, it is close enough. When I get my dev rig back I will measure it. Leant it to a friend who is having trouble with his son's.
Cheers

---- Timm Korte wrote ----

savek-cc, 2016-06-05:
Attached is a log of the power consumption (measured as current in mA with 1900 samples (@2 samples/sec))
[image]

brutevinch, 2016-06-17:
Hi there! I Found bug which make current in sleep 0.01mA. Please someone fix code!! BT have parasite power when P1_0 goes to LOW (sleep mode). BT take power from P1_6. All we need is: when wixel start sleep function we must deinitialize uart function at pin P1_6 and make him GPIO with state OUTPUT after this BT stops take power from Wixel's P1_6.

jstevensog, 2016-06-17:
Hello,
All P1_* outputs are set to output_low in the code prior to sleep.
Are you saying that P1_6 needs to be output and High?  Or would this be
better to be input?
This would be easy to do and I can implement it quickly.

Wixel and CC2511 documentation says that all outputs should be low to
prevent power drain in sleep.
Perhaps a diode in circuit to prevent this?
Cheers

brutevinch, 2016-06-17:
@jstevensog P1_6 must be OUTPUT with state LOW right after P1_0 become LOW.

jstevensog, 2016-06-17:
That is what the code does right now.
Cheers

---- brutevinch wrote ----

brutevinch, 2016-06-17:
@jstevensog How to test it: test voltage at P1_6 when BT must sleep. We need LOW state there. I didn't see 0V when bt sleeps

brutevinch, 2016-06-17:
if(sleep_ble){
  if(send_debug) {
    printf_fast("turning off BLE\r\n");
  }
  setDigitalOutput(10,LOW);
  ble_connected = 0;
}

as i understand here needed function. And we need something like this: 

if(sleep_ble){
  if(send_debug) {
    printf_fast("turning off BLE\r\n");
  }
  setDigitalOutput(10,LOW);
  setDigitalOutput(16,LOW);
  ble_connected = 0;
}

jstevensog, 2016-06-17:
The call to makeAllOutputs(LOW) in goToSleep. Sets all P1_ outputs low.
Cheers

---- brutevinch wrote ----

jstevensog, 2016-06-17:
I made that call separate, mostly for debug.  I can move that into makeAllOutputs for clarity. But I doubt this is the reason for the issue. 
Can you measure actual current drain on each pin in sleep mode?  
Cheers

---- brutevinch wrote ----

brutevinch, 2016-06-17:
In sleep mode on P1_6 current drain is 7.5mA. Here is HIGH level, Caaarl! HIGH! ))  Maybe problem is:
Peripheral functions: Many of the pins on the CC2511 can be configured to be used by a peripheral by setting the right bit in the P0SEL, P1SEL, or P2SEL register. When a pin is being used by a peripheral, the functions in this library may not work. For example, if you have enabled USART1 in Alternate Location 1, then you can not control the output value of P1_6 using these functions because P1_6 serves as the serial transmit (TX) line.

jstevensog, 2016-06-17:
OK. Will look into it over the weekend. Will complicate the sleep/wake code a bit.  All documentation says that the functions will work, bit happy to try.
Cheers

---- brutevinch wrote ----

brutevinch, 2016-06-17:
@jstevensog Thank you. I think all of us will be happy with so huge improvements.

savek-cc, 2016-06-17:
:)

jstevensog, 2016-06-17:
Hi All, particularly @brutevich,
Can someone please give this code a try and tell me if the sleep current
consumption has dropped?
I am out of my dev wixels, been giving them away/lending them out to people
in need, and don't have a spare one to try it.
Don't want to risk my Production rig, and am not set up to do the current
measurement.
Please let me know the result.
Thanks in advance.
Cheers

-- 
John Stevens
"You are how you live, not what you have."

Pololu Wixel Application - www.pololu.com
1.0
====== license
Copyright (c) 2009-2011 Pololu Corporation.  For more information, see

http://www.pololu.com/
http://forum.pololu.com/

Permission is hereby granted, free of charge, to any person
obtaining a copy of this software and associated documentation
files (the "Software"), to deal in the Software without
restriction, including without limitation the rights to use,
copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the
Software is furnished to do so, subject to the following
conditions:

The above copyright notice and this permission notice shall be
included in all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES
OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND
NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT
HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY,
WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR
OTHER DEALINGS IN THE SOFTWARE.
====== cdb
S:G$param_radio_channel$0$0({4}SL:S),D,0,0
L:G$param_radio_channel$0$0:5613
====== hex
:0404000002048D3233
:01040B0032BE
:0104130032B6
:03041B00022BE6CB
:0104230032A6
:03042B00020B8938
:010433003296
:01043B00328E
:010443003286
:01044B00327E
:010453003276
:01045B00326E
:030463000206EEA0
:01046B00325E
:03047300022BA8B1
:01047B00324E
:0304830002309CA8
:1004E60078747440F2C215C216752900752A007513
:1004F6002B00752C00C200C201D20F7800E4F2086E
:10050600F208F208F27804E4F208F208F208F27847
:1005160008E4F2783074CEF2783174D5F278327419
:10052600E6F2783374E5F27834E4F278357464F2FE
:0A053600783674C7F2783774D1F2FA
[... 1384 lines cut ...]
:1052C7006BC239C23AC23BC23CC23DC23E791D776E
:1052D70000E493A324C64017240A50157005B51D92
:1052E70002D23ED23DFAE775F00AA42AF780E224FB
:1052F7000A2430B46C04D23980D7B46804D23A8017
:10530700D0B4730302533EB464061254C702539ACF
:10531700B475061254C70253BAB4630518E6188069
:105327000DB47803025379B42E0280A56006125596
:10533700B90252C20255D3C083C08286F018868351
:1053470018868218303D19C23EC082C08312526748
:105357006007A3151DE51D70F41255A4D083D082F4
:105367001252676006A31255B980F5D082D0830226
:1053770052C2125391203A0C1253913039061253EC
:10538700911253911255CB0252C2E61255A9E61853
:105397000255AAED30E71CD23CE9F42401F9203A82
:1053A70012EAF43400FA30390AEBF43400FBECF477
:1053B7003400FC203A10303908EB7021EC701EC223
:1053C70039EA7019D23AE97014303D09E51D6005D4
:1053D700151D1255A474301255B90252C2C082C0AD
:1053E700831254E1303D60790A30391DEB54F07077
:1053F7003619EB540F703019EA54F0702A19EA5431
:105407000F702419EF54F0701E7905203A0BEF54F2
:105417000F701419EE54F0700E7903EE540F7007E5
:1054270019ED54F0700119E51DA23C994019F51DBD
:10543700C003C002C007C006C0051255A4D005D0DE
:1054470006D

JohanDegraeve, 2016-06-17:
Hi John, i have 2 spare xdrips, but it's the 'old' setup, ie without the voltage measurement resistors.
How can I see the 'sleep current consumption' ?

brutevinch, 2016-06-17:
@jstevensog My wixel not in production! But i can test all you want after 7 hours. I'm not at home

savek-cc, 2016-06-17:
@JohanDegraeve you would need to connect an amp-meter between the battery and the Wixel

JohanDegraeve, 2016-06-17:
ok I will do the test, probably late this evening ?

JohanDegraeve, 2016-06-17:
can someone compile the updated code and send me the updated wixel code ?

brutevinch, 2016-06-17:
Already compiled. Just copy it to wxl file

Отправлено с iPhone

JohanDegraeve, 2016-06-17:
is it here on github ? the last commit is from 22 days ago

jstevensog, 2016-06-17:
Hi All,
I attached the latest xBridge 2 code to treat to the last email I sent. I did not want to push this up until tested.  It is v2.43.
I'll copy it elsewhere for you to pick up.
Cheers

---- Johan Degraeve wrote ----

jstevensog, 2016-06-17:
OK.  I have put xBrideg2-test on my github repo.
Please test and let me know.
Going back to sleep for a couple of hours.  04;00 here.
Cheers

brutevinch, 2016-06-17:
@jstevensog Now P1_6 is fixed. When BT in sleep mode on P1_6 now LOW level. Good. But on P1_7 1.13V. Please make p1_7 LOW too. Now i ready to test.

JohanDegraeve, 2016-06-17:
installed on an xdrip, and connected with android device, configured the transmitter id,
but it looks like i don't receive anything at all ? the yellow led never lights up

JohanDegraeve, 2016-06-17:
by the way it pulls 33,7 mA in this situation

jstevensog, 2016-06-17:
Interesting. The code I am using sets all 8 P1 outputs to GPIO outputs low right before sleeping. I'll try and see why there is a difference, but that one really had me stumped. 
Cheers

---- brutevinch wrote ----

jstevensog, 2016-06-17:
Yellow LED will only light up on an xBridge circuit.  It indicates the BLE is connected.
I was concerned that the UART may not wake after sleeping with this change, but there is no reason why it shouldn't.
Cheers

---- Johan Degraeve wrote ----

JohanDegraeve, 2016-06-17:
I also don't see any values appearing on the app.

jstevensog, 2016-06-17:
Have you connected a serial terminal to the wixel?  You will need to ensure
it has started correctly, and is receiving packets and trying to send them
to the phone.
The only changes that have been made are in the goToSleep function to set
the type and direction of all P1 bits (change P1SEL to 0x99 and P1DIR to
0xFF) before sleep.  The code to store and restore these after sleep was
already there.  No change has been made to packet reception or transmission
code.
Cheers

jstevensog, 2016-06-17:
New version to test.  Just posted.  This should set P1_7 low.
Found a bug in makeAllOutputs that was ignoring both P1_0 and P1_7.
Cheers

JohanDegraeve, 2016-06-18:
ok now my test succeeded, maybe my transmitter id was not correctly set

It starts at 30 mA, when bg value is transmitted, then it goes to 20 mA
about 30 seconds before the next value is expected, it goes back to 30

brutevinch, 2016-06-18:
@jstevensog Now it works! Thank you. 0.1-0.2mA in sleep mode and 35-36mA in working state.

jstevensog, 2016-06-18:
Thanks Johan Degraeve,
Glad it's working. If your bridges are classic xDrip, the 20mA is mostly the HM-1x module. 
Maybe try modifying one to xBridge?
;)
Cheers

---- Johan Degraeve wrote ----

jstevensog, 2016-06-18:
Fantastic news brutevinch,
I will publish this version in the morning and inform the Facebook groups, giving you the credit for finding the bug.
Thanks so much for your work.
Cheers

---- brutevinch wrote ----

brutevinch, 2016-06-18:
@jstevensog All thanks to you. I recommend to wait 2-3 days until we hard test it.

JohanDegraeve, 2016-06-18:
@jstevensog  I assume it's only the connection Status (on HM-10) to P1_2 (on wixel) that needs to be added ?

jstevensog, 2016-06-18:
Yep, Johan, that should be all. As the code will think it is a full xDrip circuit, the battery level may be slightly out. It should be reasonable after a full charge, but I can't remember how far out the low limit will be.
It may be worth changing the voltage divider resistors as well. I'll work out the difference before I go to bed and post here to let everyone know.
Also, forgot to thank Timm who identified the higher than expected battery drain in the first place.
Sorry I never got around to thoroughly testing that aspect.
Cheers

---- Johan Degraeve wrote ----

jstevensog, 2016-06-18:
So, I have just put this new code on my production bridge.  Functioning
fine, except I believe there is a poor connection issue between the HM-1x
and wixel.  Took a while to get it's TXID from the app, even thought it was
trying and getting one sent back.  I better get out the soldering iron
tomorrow.

Johan, with the classic xDrip-wixel voltage divider resistors, the code
will "learn" your upper limit fine on the first full charge.  However, I do
not let the wixel learn too much lower a limit than the original programmed
value.  The reason being, there is a risk if the wixel is writing to flash
as the power dies, it cold corrupt the flash, and therefore the wixel
program.  I'm not sure how low you let it get, but if you are sticking with
the 2k3/1k voltage dividers, definitely charge at 40%.  It can fall to
around 35%, but I wouldn't risk it getting lower if I were you.

Cheers

jstevensog, 2016-06-18:
BTW, given that the only change to the code is in the goToSleep and
makeAllOutputs functions, it should be fine to release.  But I will leave
it until Wednesday 12;00AEST (+10 UTC) before pushing the changes and
announcing them on the Nightscout pages.

So far so good though.
Cheers

JohanDegraeve, 2016-06-18:
ok thank you all and have a good sleep :)

savek-cc, 2016-06-18:
@jstevensog : nothing to be sorry about. Thank you for all your great work in this project!

brutevinch, 2016-06-20:
@jstevensog here is a bug with getting transmitter id from app. I can't reproduce it :(

jstevensog, 2016-06-20:
Can you please connect a serial terminal and provide me with the debug
output to review?
My prod bridge initially took some time to receive a valid TXID packet from
the app.  I assumed it was to do with a poor connection between the HM-1x
TXData and the Wixel RXData pins.

As I said, the only thing in code that has changed is in the goToSleep and
makeAllOutputs functions.  Neither of these would impact receiving data
from power up.

But I will review the code just in case something else was inadvertently
changed, such as baud rates etc.
Cheers

brutevinch, 2016-06-20:
@jstevensog now all working good, i can't reproduce it. when it was it's look LIKE this (i forgot those messages)

No dex_tx_id.  Sending beacon.
29425: sending beacon Now
Sending: ▒
No dex_tx_id.  Sending beacon.
29425: sending beacon Now
Sending: ▒
No dex_tx_id.  Sending beacon.
29425: sending beacon Now
Sending: ▒

jstevensog, 2016-06-20:
Yes, that is almost what I was seeing with my production bridge, except it
was receiving the TXID packet, but just not recognising it.

Today, I just reloaded a slightly modified version of the code that I made
after your e-mail, and it worked after one failed attempt to get a packet.
I will monitor it for a period to determine if it is also working with the
ACK packets better than it was over the weekend.

It is strange behaviour I have seen on a couple of occasions.  As it varies
so much between loads of the program, I am not sure I can counter it in
code.
I really need to get my Dev rig back so I have something to work with.
Thanks for the heads up.
Cheers

jstevensog, 2016-06-20:
I just realised, your issue may also have been due to the phone's GATT
service.  I have found on occasion the GATT "locks up" and stops responding
to messages from the bridge.  It requires turning Bluetooth off and on, or
restarting the collection service on xDrip, or just waiting until it starts
working again.

The output you sent me looks more like that issue, and you cannot repeat
it.  It either happens or it doesn't.

I have altered a version of xDrip to try and work around this.  The current
DexCollectionService module in xDrip closes the GATT on every disconnect.
This is apparently the recommended way of dealing with the GATT when a
device disconnects, but some Android gurus have said this is a bad idea.
Given we disconnect every 5 minutes (assuming we get a packet from the
bridge, and acknowledge it), this is pretty regular.
I modified DexCollectionService to only FULLY close the GATT one in every 6
packet captures.  So, keep it around for 6 captures, then close and
re-open, effectively meaning 30 minutes or more between GATT closures.
So far I have not needed to turn off and on Bluetooth on my phone since,
but I believe it is still early days.  If this proves a reasonable solution
for stability, I will ask it to be merged into the xDrip-Experimental
master and it will end up in a future release.
Cheers

savek-cc, 2016-06-20:
I also have that issue where I have to turn Bluetooth off and back on again on the phone to continue receiving packets. Couldn't make out a pattern - sometimes it happens twice in one hour, sometimes I get almost a day without this problem. As an app I'm using xDrip+ though (https://github.com/jamorham/xDrip-plus).

brutevinch, 2016-06-20:
@jstevensog also using xDrip+

jstevensog, 2016-06-20:
As xDrip+ is based on xDrip, I am certain the issue is the same.  If my
"fix" to DexCollectionService proves valuable, I will let @jmorham know so
he can put it in xDrip+.  The issue is a known one with the Android GATT,
and is different on every version of Android.  There is a abstraction
library that is supposed to deal with all version, but I am not sure it has
been tested that well yet.
Cheers

jstevensog, 2016-06-20:
Wow. I may actually have to charge my Pebble more often than my bridge. When I fully charge my bridge in the past, it used to get below 100% in an hour or so. Unless I have somehow broken my voltage divider again, it has been half a day and still at 100%.
Cheers

---- brutevinch wrote ----

brutevinch, 2016-06-20:
@jstevensog Yeah! i think with bl5c it must be ~ 1 week =) And yes maybe we can reduce consumption more) <- it was teaser) When you fix current nuance we discuss it.

jstevensog, 2016-06-21:
Hi all,
I am very pleased with v2.43. I will release and announce tomorrow, in roughly 6 hours, unless anyone has a serious issue why not to.
Thanks all for your help.
Cheers

---- brutevinch wrote ----

JohanDegraeve, 2016-06-21:
I have a question, not specific for this release but in general. What happens with the bluetooth connection if the bluetooth module goes in sleep mode ?
Does the app see it as a lost connection ?

I'm asking because I'm trying to develop an iOS app. iOS seems to put the app in suspended mode (ie not active anymore) when the bluetooth connection is lost - still testing but the xDrip setup. I still need to add the wire to have the xBridge setup.

jstevensog, 2016-06-21:
In xDrip DexCollectionService, this is a GATT disconnect event. The GATT is closed, and then reconnected a short time later, with periodic attempts to get a connection.  The DexCollectionService continues to run regardless. It only stops when it is specifically told to do so.

xBridge2 has a terminal command you can set to keep the BLE on and connected, buy it does consume a lot of battery that way. I put it there for those folk whose phones Android implementation doesn't like the disconnections as much as the better implementations do.  I have found that regardless of this setting, I still have to occasionally cycle Bluetooth to get xDrip to recieve packets, as many have also reported for the original wixel-xDrip.
Cheers
Cheers

---- Johan Degraeve wrote ----

JohanDegraeve, 2016-06-21:
If the bluetooth is in sleep mode, what's the status of the red led ? continuously on, or out or blinking ?

jstevensog, 2016-06-21:
Johann, there is no status on the wixel in sleep mode. All LEDs are off, all inputs set to outputs, all outputs set to high impedance. This turns off the HM-1x, so no status there either.
If you have set the bridge to keep the HM-1x on during sleep, the HM-1x status LED will (should, I haven't tested it) remain on while it is connected to the phone.
Cheers

---- Johan Degraeve wrote ----

[11 more replies]

closed 2016-06-30 by jstevensog