jstevensog/wixel-sdk #23: waitDoingServices does not work with break flag

savek-cc, 2016-06-25:
Giving the break_flag to waitDoingServices apparently doesn't have any effect - it will always wait for the full timeout. Passed by value? Passing by reference is apparently not possible with SDCC.

jstevensog, 2016-06-27:
Hi Timm,
Yeah, I found out a while ago that this is the case.  I should really
change it.
I really want to rewrite xBridge completely soon.  It is not original code
by me and the many contributors have done a great job, but all very
differently.
All the comments in there are either from original code, or added by me as
a learned how it worked.  I am a stickler for lots of comments in open
source code.
I never assume that someone can read the code without comments to
understand what is actually going on.
So, once channel scanning capture is working and in the code, I will do a
full rewrite and change the operation to a state machine format which will
be much more efficient.
Means more global booleans and stuff, but will be much better.
Cheers

savek-cc, 2016-06-27:
Hi John,
in my repository, I replaced the waitDoingServices function with a function macro - this now actually does resprect the break_flag and works just like the function otherwise - maybe it could be interesting for you to include 'till you get around to a rewrite ;)
Timm

jstevensog, 2016-06-27:
Hi Timm,
Will definitely get this into the public code. Thanks for this. I never got around to solving that issue.
Cheers

---- Timm Korte wrote ----

savek-cc, 2017-05-14:
Guess this issue can be closed, as you included the code in https://github.com/jstevensog/wixel-sdk/commit/477f6e1a25755683861997f66447550eab65139c

closed 2017-05-14 by savek-cc