jstevensog/wixel-sdk #27: HM11 range issue

Nemo1985, 2017-09-08:
Good morning, I assembled the xbridge, my version is like the Chris Bothello’s HM-11 board, just with a different battery (Lithium Ion Cylindrical Battery - 3.7v 2200mAh), my main issue is the range of the bluetooth module, even if I keep it like less than a meter far from me, mostly when I'm in bed it always miss readings, last night it missed reading fo like an hour.
Is there any way to increase the range?
I also have another problem, sometimes the red led of wixel keeps blinking and my battery stand much less than it should.

Do you have any advice for me?

Thank you in advance.

jstevensog, 2017-09-08:
Hi,
On range of the BLE module, there is little that can be done.  But it
sounds more like you are saying that the wixel range (your transmitter to
the bridge) is not so good.  Does this happen only at night, or is it all
the time?  If only at night, do you have an electric blanket, or any other
potential interfering or signal blocking things between the transmitter and
your bridge?

My bridge is mostly in my pocket.  At night, it could be up to 10 feet from
me, and still picks up most packets from my transmitter.

The wixel RED LED should always blink while it is awake and waiting for a
packet.  You can turn off all wixel LEDs by connecting your wixel to a
serial terminal program (PuTTY or KiTTY), waiting for it to show output,
and hitting the L key.  This key toggles the LEDs on and off.  You will see
a message saying so.  This is explained in the xBridge2 document.  Please
read the document and follow the instructions in there.  It is important to
know what all the LED indications mean, and all the command options.

I leave the LEDs turned off all the time on my production bridge.

Hope this helps.  I am interested in your response on if the range issue is
a problem only at night or not.
Cheers

Nemo1985, 2017-09-09:
Thank you for the answer.
The problem is more common when i'm in bed, I don't use an electric blanket, I just have a cable that I use to charge my phone.
Last night I kept the bridge almost next to me but for like an hour it lost the signal (while the official Dexcom g4 had no issue), I always keep them in almost same position, I start to think that the problem could be that I sleep on stomach position.
https://www.uploadhouse.com/viewfile.php?id=25425912
https://www.uploadhouse.com/viewfile.php?id=25425919
That being said, in general the signal seems weaker.

Thank you for the advice for the leds, I used putty to turn them off.

Nemo1985, 2017-09-28:
I'm sorry if I post again, I updated to version 2.47e and the problems are getting bigger, the worst thing is that I'm unable to find the culprit.
Now the xbridge usually lose the signal 3-4 at days, to restore the connection I have to restart it and find it again with the phone.
The weird thing is that the phone says connected even when the signal is lost, so i'm thinking that the issue should be on my xbridge by itself. When I connect using the mini usb port and watch the output on putty it usually stop respond soon enough...

jstevensog, 2017-09-29:
Hi,
Don't apologise for posting, I would like to help.  All reports of v2.47e
prior to release and since have been very positive, so I am not sure why
you are having bigger problems.
When you say you lose signal 3-4 days, do you mean that it works well for
3-4 days and then it stops?
What do you mean you have to restart it and find it again with the phone?
You power cycle the bridge? You must scan for it in xDrip+?  You should not
have to do this at all.  xDrip+ will always remember the bridge, until you
make it forget it.

Some phones and versions of Android still have issues with Bluetooth and
BLE over time.  My own phone will eventually stop receiving data from my
xBridge, even though it is connected.  It is often worse after app
updates.  I am now in the habit of restarting the phone once I do app
updates.  Since doing that, I have had no issues.

Please tell me what type of phone you are using and what version of Android
it is running?  I can do no more on the xBridge side.

If I understand you correctly, when you plug the bridge into a computer USB
and run PuTTY, it stops giving output when it goes to sleep.  That is quite
common, and I am not sure why.  I believe it is either a hardware issue, or
an issue with the wixel driver.  My own laptop has one USB port (USB3) that
works fine when the bridge sleeps, but the other (USB2) does not.  Both
used to work fine, but at some point after windows updates, the other
stopped working.  I can only suggest trying another USB port or computer.

I am sorry you are having these issues, but I need to understand what you
are doing.  It may be that the wixel is faulty.  Do you have a second one
to try?
Cheers

Nemo1985, 2017-09-29:
Thank you for you answer and sorry for my bad english.
I mean that 3-4 times at day it just suddenly stop to receive the readings.
I have to stop the wixel (I unplug the battery cable) and from the application (xdrip+ lastest stable, should I switch to the lastest nightly build?) I do a new bluetooth scan and pair it again (I do this second step just to be sure, just restart the wixel could be enough).
Then it works for some hours, they can be 5 or just somet minutes.
My phone is a Xiaomi Redmi note 4 with android version 6.0.1.
Next week I will try to sold the wires again from scratch.
I usually use the usb 3 of my computer (it's quite new and I have just 2 usb 2.0 ports on the fron panel).
This is the full output from putty until it stopped to answer (this time I tried to connect it to my pc without using the battery):
```
Starting xBridge v2.47e
Retrieving Settings
11023 - send_data AT (2)
Sending: AT
Response: OK11064 - got OK
11064 - atBt() got OK
11064 - send_data AT (2)
11065 - send_data blocked on uart1 input
Sending: AT
Response: OK11086 - got OK
11086 - atBt() got OK
batteryPercent val: 363
batteryPercent returning 0
looking for 6907045 (6JR55)
12094: last_channel is 0, delay is 0
12095 - start is 12095, waiting for packet on channel 0 for 0
OK+CONNble connected
got a packet at 139878 on channel 0
139878 - got pkt
```
batteryPercent val: 364
batteryPercent returning 0
139880 - packet waiting
139880 - send pkt
139880 - send_data  (17)
139880 - send_data blocked on uart1 input
Sending: ▒.▒▒di
Response: 139897 - waiting for ack
▒139978 - Processing ACK packet
139978 - UART1 xBridge protocol done
140479 - sleeping for 260
140479 - sleep_time_ms is 250000, this_sleep_time is 9399
```
I have received just 1 reading and then it stopped, on bluetooth settings I have checked those 3 options:
Tunr Bluetooth on
Bluetooth Watchdog
Close GATT on BLE disconnect

Thank you for your help and patience.

jstevensog, 2017-09-30:
Hi,
Again, please do not apologise for your level of English.  It is way better
than my level of your language.  I hope I am making my English easy for you
to understand.

I need to understand what is happening is happening with the bridge.
Please make sure you have LEDs enabled.

When you have the "black out", does the bridge stay awake?  Or does it go
to sleep?

When things are normal, you should see the following sequence:
1) When the bridge wakes up, it should have a flashing RED LED.
2) When the bridge connects to the phone, it should have a solid YELLOW LED.
3) When the packet is captured (about 40 seconds after waking), the RED LED
will either go off or go solid on.
4) If the phone accepts the packet, it sends an ACK, and the bridge will go
to sleep.  If the ACK is not received by the bridge, it will go to sleep
after 2 minutes.

It will repeat this cycle roughly every 5 minutes.

When you get a "black out" please note what the LEDs are doing.  If the
bridge is staying awake for longer than 5 minutes, or staying awake
permanently, it is not receiving packets.  This will likely be due to a
faulty wixel.  When you disconnect the battery, and reconnect it, the wixel
radio is reset.

If the wixel goes to sleep after 2 minutes once it indicates it has
captured a packet, it means that the phone is the issue.  The phone's BLE
stack has become unstable.  In this case, when you forget and reconnect the
bridge, it clears the BLE stack.  Please try turning the phone's Bluetooth
off and on and see if that stops the "black out".

The Xaomi phones have proven to be somewhat unreliable on the BLE side.  My
own Sony Xperia Z3 can also have similar issues, which I have learned to
avoid by regularly powering off and powering on the phone, particularly
after app updates.
I hope this helps.

v2.47e now ensures the HM-11 board breaks the BLE connection before power
down.  This has seen a great improvement in BLE black outs for everyone
that beta tested it.

Cheers

jstevensog, 2017-10-01:
Hello,
Today is the first day I experienced something similar to what you have,
but only twice.  The only solution that worked was power cycling the
bridge.  I believe this may be due to temperature affecting either the
bridge (more likely) or the Transmitter.  On the receipt of each packet, if
there is a frequency difference between the frequency the receiver is set
to, and the frequency the packet was actually received.  This difference is
used to tune the receiver.

There may be a chance that a large temperature change, seems to be cool to
warm, that upsets this.  I will try to adjust the code to reset the
receiver tuning IF a packet is missed.  Give me a day or two to figure out
how this is best done, make the change, and then test (if I can).
Cheers

Nemo1985, 2017-10-01:
Hello, 
sorry for the late answer, during the last 2 days, I kept the bridge connected to the pc because a wire from the wixel to the Adafruit Micro Lipo w/MicroUSB Jack has unsoldered.
What I saw is: with the 2.47d version I have no issues (I have readings for hours), the readings are ok, while with the last version it stops working after the first reading.
I think I may have an issue with the led, because the red led is always on, even if I turn off from the leds from putty, so I fear I can't give you any indication from the leds (this happen both with 2.47d and e).
The led issue could be caused from some tin gone to another hole?
If I can help you with any test I will do my best.

jstevensog, 2017-10-11:
Hi Nemo1985,
The LED issues is something I cannot advise on.  I would use a magnifier to
look for any solder bridges on your wixel.  That is all I can think of that
this would be.  Knowing the status from the LEDs is very important to
determine what is going on.

It does not make sense that v2.47e stops working after the first capture,
but v2.47d works fine.  They are essentially the same code.  The only
difference is v2.47e adds support for the HM-16/17 modules, and uses the
same code for HM-10/11s which I have tested and works fine.

I am assuming that you are running both versions on the same wixel.  Is
that the case?  If so, what stops working?  Is it only the output to the
terminal (very common) and the wixel still captures packets that are seen
in xDrip+?  Or does the wixel never capture a packet after the first one?

Attached is version 2.47f.  This version has some additional code to reset
a channel's offset to default if the channel times out listening for a
packet.  Also, it tries to make sure that a detected channel offset on a
good packet is sane before it uses it, otherwise it leaves it alone.

Please try this version and let me know how it goes.
Cheers

-- 
John Stevens
"You are how you live, not what you have."

Pololu Wixel Application - www.pololu.com
1.0
====== license
Copyright (c) 2009-2011 Pololu Corporation.  For more information, see

http://www.pololu.com/
http://forum.pololu.com/

Permission is hereby granted, free of charge, to any person
obtaining a copy of this software and associated documentation
files (the "Software"), to deal in the Software without
restriction, including without limitation the rights to use,
copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the
Software is furnished to do so, subject to the following
conditions:

The above copyright notice and this permission notice shall be
included in all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES
OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND
NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT
HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY,
WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR
OTHER DEALINGS IN THE SOFTWARE.
====== cdb
S:G$param_radio_channel$0$0({4}SL:S),D,0,0
L:G$param_radio_channel$0$0:6373
====== hex
:0404000002048D3233
:01040B0032BE
:0104130032B6
:03041B0002379C09
:0104230032A6
:03042B00020B18A9
:010433003296
:01043B00328E
:010443003286
:01044B00327E
[... 1327 lines cut ...]
:104AE5001D8F828E83E0FA8B828C83F0A3AB82AC20
:104AF5008380EAD22922AF82E4CF25E0CF33FE744A
:104B0500202FFF74DE3EFE78A9E2FD78AAE2FB08BD
:104B1500E2FCED60141D8B828C83E0FAA3AB82ACC2
:104B2500838F828E83EAF080E922C22890DE0174A9
:104B350001F090DE0B7407F0221252274014AFBE2D
:104B450053077F8FBE12523090F5F7E4F0C228026A
:104B55004D8790F5F7E0FF701312523843BE800081
:104B650000000090F5F77402F0124B2F90DE06E07E
:104B7500FF90DE02E0FEEF30E002D228EF30E216D1
:104B850090F5F77404F090F5F8E4F0C007C006124C
:104B95004B2FD006D007EF30E102C228EE20E0030C
:104BA500024D8790DE0EE4F090DE11E0FFD229EF92
:104BB50030E40B90DE117480F090F5F8E4F0EF30FE
:104BC500E20990DE11E4F090F5F8F0EF20E0030241
:104BD5004D1D90F5F8E0FEBE01028003024C7F906A
:104BE500DE16E0FE90F565E0FCA3E0FD8E027B009D
:104BF500C3EC9AED9B50028C0678A6EEF278A47968
:104C0500A7E2F308E209F3758200C007C006124A5D
:104C1500C6D006D00778A4E22EF208E23400F2905E
:104C2500F565E0FCA3E0FD7B0090F565ECC39EF027
:104C3500ED9BA3F090F565E0FDA3E0FE4D600990C6
:104C4500DE117440F0024D1D90F55DE05460600783
:104C5500C00712487FD00790DE0EE4F090F5F8E02B
:104C6500FE70