jstevensog/wixel-sdk #29: Lock on 4.47e

danpowell88, 2017-10-10:
Had an instance of 2.47e where I'm not getting any data.

Using a hm-10, restarting phone has no effect and BT led indicates there is a bluetooth connection.

Logs as follows

> sProcessing Status Command
> xBridge v2.47e
> dex_tx_id: 6840542 (6GQ6X)
> initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, sen                                                      d_debug: 1, do_leds: 1
> dex_tx_id_set: 1, got_packet: 0
> battery_capacity: 81
> current ms: 512082207

> sProcessing Status Command
> xBridge v2.47e
> dex_tx_id: 6840542 (6GQ6X)
> initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, sen                                                      d_debug: 1, do_leds: 1
> dex_tx_id_set: 1, got_packet: 0
> battery_capacity: 81
> current ms: 512144307

> sProcessing Status Command
> xBridge v2.47e
> dex_tx_id: 6840542 (6GQ6X)
> initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, sen                                                      d_debug: 1, do_leds: 1
> dex_tx_id_set: 1, got_packet: 0
> battery_capacity: 81
> current ms: 512253643
> 512342872 - start is 512342872, waiting for packet on channel 1 for 500
> 512343372 - start is 512343372, waiting for packet on channel 2 for 500
> 512343872 - start is 512343872, waiting for packet on channel 3 for 500
> 512344372 - missed a packet
> 512344372: sending beacon Now
> 512344372 - send_data ▒▒`h (7)
> 512344372 - send_data blocked on uart1 input
> Sending: ▒▒`h
> Response: 512344378 - start is 512344378, waiting for packet on channel 0 for 29                                                      8510

> sProcessing Status Command
> xBridge v2.47e
> dex_tx_id: 6840542 (6GQ6X)
> initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
> dex_tx_id_set: 1, got_packet: 0
> battery_capacity: 81
> current ms: 512463366

It seems to never be getting the packet and continues to be able to respond to status commands so it's not going into sleep.

danpowell88, 2017-10-10:
> sProcessing Status Command
> xBridge v2.47e
> dex_tx_id: 6840542 (6GQ6X)
> initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
> dex_tx_id_set: 1, got_packet: 0
> battery_capacity: 81
> current ms: 512688279

> sProcessing Status Command
> xBridge v2.47e
> dex_tx_id: 6840542 (6GQ6X)
> initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
> dex_tx_id_set: 1, got_packet: 0
> battery_capacity: 81
> current ms: 512689147

> sProcessing Status Command
> xBridge v2.47e
> dex_tx_id: 6840542 (6GQ6X)
> initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
> dex_tx_id_set: 1, got_packet: 0
> battery_capacity: 81
> current ms: 512860664
> 512942903 - start is 512942903, waiting for packet on channel 1 for 500
> 512943403 - start is 512943403, waiting for packet on channel 2 for 500
> 512943903 - start is 512943903, waiting for packet on channel 3 for 500
> 512944403 - missed a packet
> 512944403: sending beacon Now
> 512944403 - send_data ▒▒`h (7)
> 512944403 - send_data blocked on uart1 input
> Sending: ▒▒`h
> Response: 512944409 - start is 512944409, waiting for packet on channel 0 for 298510

> sProcessing Status Command
> xBridge v2.47e
> dex_tx_id: 6840542 (6GQ6X)
> initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
> dex_tx_id_set: 1, got_packet: 0
> battery_capacity: 81
> current ms: 513202178
>

danpowell88, 2017-10-10:
It eventually came good shortly after but interestingly even though the phone received the data the BT connection was not closed automatically

> got a packet at 513242913 on channel 0
> 513242913 - got pkt
> batteryPercent val: 1723
> batteryPercent returning 77.136
> 513242915 - packet waiting
> 513242915 - send pkt
> 513242915 - send_data  (17)
> 513242915 - send_data blocked on uart1 input
> Sending: @▒▒M▒`h
> Response: 513242932 - waiting for ack
> 513252942 - packet waiting
> 513252942 - send pkt
> 513252943 - send_data  (17)
> 513252943 - send_data blocked on uart1 input
> Sending: @▒▒M▒`h
> Response: 513252959 - waiting for ack
> 513262969 - packet waiting
> 513262970 - send pkt
> 513262970 - send_data  (17)
> 513262970 - send_data blocked on uart1 input
> Sending: @▒▒M▒`h
> Response: 513262987 - waiting for ack
> 513272997 - packet waiting
> 513272997 - send pkt
> 513272997 - send_data  (17)
> 513272998 - send_data blocked on uart1 input
> Sending: @▒▒M▒`h
> Response: 513273014 - waiting for ack
> 513283024 - packet waiting
> 513283025 - send pkt
> 513283025 - send_data  (17)
> 513283025 - send_data blocked on uart1 input
> Sending: @▒▒M▒`h
> Response: 513283041 - waiting for ack
> 513293052 - packet waiting
> 513293052 - send pkt
> 513293052 - send_data  (17)
> 513293053 - send_data blocked on uart1 input
> Sending: @▒▒M▒`h
> Response: 513293069 - waiting for ack
> 513303079 - packet waiting
> 513303080 - send pkt
> 513303080 - send_data  (17)
> 513303080 - send_data blocked on uart1 input
> Sending: @▒▒M▒`h
> Response: 513303096 - waiting for ack
> 513313107 - packet waiting
> 513313107 - send pkt
> 513313107 - send_data  (17)
> 513313108 - send_data blocked on uart1 input
> Sending: @▒▒M▒`h
> Response: 513313124 - waiting for ack
> 513323134 - packet waiting
> 513323135 - send pkt
> 513323135 - send_data  (17)
> 513323135 - send_data blocked on uart1 input
> Sending: @▒▒M▒`h
> Response: 513323151 - waiting for ack
> 513333162 - packet waiting
> 513333162 - send pkt
> 513333162 - send_data  (17)
> 513333163 - send_data blocked on uart1 input
> Sending: @▒▒M▒`h
> Response: 513333179 - waiting for ack
> 513343189 - packet waiting
> 513343190 - send pkt
> 513343190 - send_data  (17)
> 513343190 - send_data blocked on uart1 input
> Sending: @▒▒M▒`h
> Response: 513343206 - waiting for ack
> 513353217 - packet waiting
> 513353217 - send pkt
> 513353217 - send_data  (17)
> 513353218 - send_data blocked on uart1 input
> Sending: @▒▒M▒`h
> Response: 513353234 - waiting for ack
> 513363244 - No ACK received
> 513363746 - ble_connected = 1, P1_2 = 1
> 513363746 - send_data AT (2)
> 513363746 - send_data blocked on uart1 input
> Sending: AT
> Response: 513368767 - breakBt() Did not disconnect
> connection did not break, P1_2 is 0
> 513368768 - ble_connected = 1, P1_2 = 0
> 513368768 - send_data AT (2)
> 513368769 - send_data blocked on uart1 input
> Sending: AT
> Response: 513373789 - breakBt() Did not disconnect
> connection did not break, P1_2 is 0
> 513373790 - ble_connected = 1, P1_2 = 0
> 513373791 - send_data AT (2)
> 513373791 - send_data blocked on uart1 input
> Sending: AT
> Response: 513378812 - breakBt() Did not disconnect
> connection did not break, P1_2 is 0
> turning off BLE
> 513378812 - sleeping for 260
> 513378813 - sleep_time_ms is 120000, this_sleep_time is 4100
> this_sleep_time is 4100, so skipping this iteration.
> 513378814 - sleep_time_ms is 110000, this_sleep_time is 10000

jstevensog, 2017-10-10:
Hi Daniel,
I have had two instances of this occurring as well.  On the same day.  I
believe it may be a situation where the radio auto tuning code is in error
and am working of a fix.  Seems to occur when an extreme temperature change
takes place, which is what as me looking at that.
I have a version that may fix it, buy haven't had a chance to test it.  I
also want to add a return to default channel tuning when a packet is
missed, as the defaults seem to work well in picking up the signal again.
Cheers

jstevensog, 2017-10-10:
Hmmm.  that one is interesting.  I will try and replicate it.
Cheers

danpowell88, 2017-10-11:
The termperature change would probably explain this one, it occured around the time I left a very cool air-con office to go for a walk outside in 30ish degree heat and the bridge was in my back pocket

jstevensog, 2017-10-11:
Hi Daniel,
I have pushed a new version to the v2.47f  branch in my repo, and attached
the .wxl file here.  I have added two new pieces of code to the v2.47e.
1) Reset a channel's offset to default if it times out.
2) Ensure that a channel's offset is only updated with the detected
difference IF the value makes sense.

I will run this version on my production bridge for a while to see if it
helps.  Would be great if you can run it also.

It is always fun when you fix a bunch of stuff, and that uncovers other
things that need fixing. ;)
Cheers

-- 
John Stevens
"You are how you live, not what you have."

Pololu Wixel Application - www.pololu.com
1.0
====== license
Copyright (c) 2009-2011 Pololu Corporation.  For more information, see

http://www.pololu.com/
http://forum.pololu.com/

Permission is hereby granted, free of charge, to any person
obtaining a copy of this software and associated documentation
files (the "Software"), to deal in the Software without
restriction, including without limitation the rights to use,
copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the
Software is furnished to do so, subject to the following
conditions:

The above copyright notice and this permission notice shall be
included in all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES
OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND
NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT
HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY,
WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR
OTHER DEALINGS IN THE SOFTWARE.
====== cdb
S:G$param_radio_channel$0$0({4}SL:S),D,0,0
L:G$param_radio_channel$0$0:6373
====== hex
:0404000002048D3233
:01040B0032BE
:0104130032B6
:03041B0002379C09
:0104230032A6
:03042B00020B18A9
:010433003296
:01043B00328E
:010443003286
:01044B00327E
:010453003276
:01045B00326E
:030463000206ECA2
:01046B00325E
:0304730002375EEF
:01047B00324E
:03048300023D1621
:1004E60078517440F2C213C214752900752A00753A
:1004F6002B00752C00C200C201D20DD2107800E488
:10050600F208F208F208F27804E4F208F208F208B7
[... 1342 lines cut ...]
:104C750048F090F5F8E4F0024D1D90DE16E0FEBE1A
:104C850008028003024D1778A7745DF20874F5F2E7
:104C950078A67408F2758200C007124AC6D007903C
:104CA500F5F8E4F090F55DE054607009C007124D29
:104CB50088D0078007C0071247F5D00790DE0EE4BD
:104CC500F090F563E0FDA3E0FE90F565E0FBA3E061
:104CD500FCC3ED9BEE9C500890F565EDF0A3EEF05E
:104CE50090F5F8E0FE700890DE117460F08029BE42
:104CF500012690F565E0FDA3E0FE4D600890DE110C
:104D05007440F0801390DE11744AF090F5F8E4F0E9
:104D1500800690DE117440F0EF20E16690F5F8E032
:104D2500FEBE025E90F565E0FDA3E0FEC3ED9420B6
:104D3500EE940050097F0A90F5F8E4F080047F02B4
:104D45007D2078A9EDF278A479AAE2F308E209F3C7
:104D5500758200C007C005124AFBD005D00790DE5A
:104D650011EFF078A4E22DF208E23400F290F56537
:104D7500E0FEA3E0FF7C0090F565EEC39DF0EF9C9F
:104D8500A3F02290F567E4F090F568F090F55EE009
:104D9500FF24F35003025021EF2F2F904DA47302EF
:104DA5004F4B02500D02502102500D025021024E70
:104DB50086024DCB025021024EE4024EA7024F0956
:104DC50002502102501290F55FE0FEA3E0FFFC7D4A
:104DD50000BC0105BD00028013BC0205BD000280B8
:104DE50020BC0305BD0002803B024E7078A47427E9
:10

danpowell88, 2017-10-11:
Will do, hard to verify that its not happening, only happened to me that once but given its starting to warm up here I might potentially see it more often

jstevensog, 2017-10-11:
Yeah, it has only happened to me twice on one day.  Cold outside, warm
inside.  Whenever I spent extended time outside and came back inside, it
seemed to occur.  Had to be in the cold for a very long time though.

So far v2.47f is working just as well as v2.47e.  Will go out riding today
(18C top today) and see what happens when I get home.
Cheers

danpowell88, 2017-10-12:
Yeah overnight I had no issues

danpowell88, 2017-10-18:
Still no issues, can't say if it fixed the problem explicitly though

ToreBj, 2017-10-19:
I've been watching this conversation and I forked the Xbridge2 a year ago approx. and made a review of the code and tried to improve the battery supervision among a few things. Doing that I noticed though that you are in general not handling rollover of timers/counters. I'm not sure of the effect in each case in the different codeparts. Is this something you are aware of and have considered? I think this may affect the code execution after some time(weeks?)

jstevensog, 2017-10-19:
Hi Tore,
Yes, I am aware that I am not dealing with rolling over of timers/counters
everywhere.  In most instances, this is not an issue.  But in the few cases
I believe this is an issue (ms count for example, used to calculate the
sleep time) I have, I believe, dealt with them correctly.
Though, I am happy to be corrected, and happy for others to contribute to
the code.

Most of it is fairly untouched from the original code base from Adrien de
Croy, who in turn left it basically untouched from his original source.
But there a definitely areas that need improving.  So many contributors,
each with their own coding styles/standards, does make it difficult to
follow.  I am planning a full rewrite of the code for xBridge3, but that is
still a while away.

So, if you wish to contribute, please feel free.  I am happy for any
assistance and review of the code, and happy to give credit.

The issues we are discussing are since the sleep timing and initiation was
improved and the code became much more stable and better at capturing
packets.  The most common state when the lock up we are discussing occurs
is that the bridge is awake and connected, but fails to receive packets.
There are two possibilities.

1) The receiver auto tuning enters a very incorrect value to the offset for
each channel.  When this occurs, no packet is captured on any channel, and
the bridge stays awake.  I have added code to reset the offsets to defaults
when packets are missed on a channel.
This can easily happen because the offsets by default are large negative
values, I believe because the original coder used this as a method of
tuning to any transmitter drift.  I plan to go through the frequency
calculations for each channel to see if I can improve this and get these
default offsets closer to 0.

2) There is a place in the code going into an infinite loop.  I have caused
these before and need to go through the code again to try and find out if
this is the case.  Would be happy if you have time to review the latest
v2.47f code and see if you can point out potential code areas that this may
be happening.

Another possibility is that on occasion the bridge fails to detect the BLE
connection.  It doesn't bother to send the packet on if it doesn't detect
BLE is connected, but it should also go to sleep after 2 minutes.  Unless,
of course, I have a bug in the code that doesn't permit this, and it is
essentially an infinite loop waiting for the connection.

Anyway, that is my thinking on this very intermittent occurrence of a lock
up.   If you, or anyone, has any suggestions I would be most interested and
appreciative.
Cheers

ToreBj, 2017-10-19:
Hi John,

I will make a code diff (against the version I've been using: 2.46) and check your changes and see what you are trying to achieve. I'm currently using a G5, but I've got a working G4. I have the standard HM10/Wixel/Adafruit w /battery monitoring "setup-in-a-box" and 2 of he Xdrip Lipo board(one w/HM11 and one w/HM17) in a 3D printed case. I think and have seen different issues with BLE when testing with different phones. HM17 behaves differently when already bonded and requires a different setup as I can recall. I'll try to dig into my memory and the code to see how I can contribute.

jstevensog, 2017-10-20:
Thanks Tore,
You would be better to do a pull of v2.47f branch to begin with, and maybe
then do a diff between v2.46.  Quite a lot has changed in v2.47e/f around
how the cc2511 is put to sleep, and how quickly it wakes up.  It is now
waking every 10 seconds to sync the RCOSC and HSXOSC for more stable and
accurate sleeping.

v2.47e/f also no longer uses the STATUS wire.  The HM-17s do not set this
correctly to a LOW/HIGH level for some reason, so I have moved to setting
the HM-1x modules to output OK+CONN/OK+LOST when a connection is
established/broken, and send an AT to break the connection before putting
the wixel to sleep, as keencave does in blucon.  This alone has
SIGNIFICANTLY increased the BLE reliability on the phone side.  It also
means no difference in circuitry with the older wixel-xdrip circuit, other
than powering off the HM-1x to save power.

I am running v2.47f on my prod bridge now and easily achieve 7 days between
charges with a 1200mAh battery.  Best it has ever done, and it still uses
an old HM-10 module.  I have a dev bridge with a HM-17, and intend to try
it out in prod once I get hold of a board I can solder it to.

Any assistance is greatly appreciated.
Cheers

danpowell88, 2017-10-20:
I think I managed to get the same issue again, currently ~20 mins without any readings

> sProcessing Status Command
> xBridge v2.47f
> dex_tx_id: 6840542 (6GQ6X)
> initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
> dex_tx_id_set: 1, got_packet: 0
> battery_capacity: 77
> current ms: 426796801
> sProcessing Status Command
> xBridge v2.47f
> dex_tx_id: 6840542 (6GQ6X)
> initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
> dex_tx_id_set: 1, got_packet: 0
> battery_capacity: 77
> current ms: 426948875
> 427064834 - start is 427064834, waiting for packet on channel 1 for 500
> 427065334 - start is 427065334, waiting for packet on channel 2 for 500
> 427065834 - start is 427065834, waiting for packet on channel 3 for 500
> 427066334 - missed a packet, resetting channel offset to default
> 427066334: sending beacon Now
> 427066334 - send_data ▒▒`h (7)
> 427066335 - send_data blocked on uart1 input
> Sending: ▒▒`h
> Response: 427066341 - start is 427066341, waiting for packet on channel 0 for 298510

This time I'd been inside for a number of hours and the bridge was sitting in the same spot in a cool air-con office.