jstevensog/wixel-sdk #32: 2.47f problems

danpowell88, 2017-10-24:
I seem to be getting a lot of missed packets on 2.47f, it usually happens after being powered on for some time and then will drop off.

Watching the logs I can see that its not even doing any activity at all

```
sProcessing Status Command
xBridge v2.47f
dex_tx_id: 6840542 (6GQ6X)
initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
dex_tx_id_set: 1, got_packet: 0
battery_capacity: 93
current ms: 10642037
```

This is against the latest 2.47f wixel file.

Restarting does seem to fix it temporarily

jstevensog, 2017-10-26:
Hi Daniel,
Sorry for the delay in replying.
This is very strange indeed.  I have 2.47f running on 3 different bridges,
one being my prod bridge and it has been even more solid than v2.47e on
packet capture, and absolutely no bridge lock up since I put it in
service.

It might be that I have inadvertently "tuned" the defaults to my
transmitter, so I will try and test two of the bridges with another
transmitter.  But that really shouldn't be the case.  I have adjusted the
first channel carrier frequency only marginally to bring the default
channel offsets closer to 0.  The first packet on each channel captured
from my prod transmitter shows the adjustment just fine, and from then on
tracks very well.  Of course, this is mostly channel 0.

I just checked that the .wxl in the v2.47f branch is correct, so I haven't
mistakenly uploaded a potential dud version.

Will try and check this out today.
Cheers

danpowell88, 2017-10-27:
Yeah was really weird, I have since switched back to e and everything seems fine, I had to delete and reinstall xdrip completly due to some other issues but I doubt that would effect the scanning stuff seen here.

Another thing that may just be due to my phone is it seems the BT connection isn't being closed as  quickly.

old-square-eyes, 2018-02-11:
I think I'm also getting this issue. https://github.com/jstevensog/wixel-sdk/issues/37

danpowell88, 2018-02-11:
I had the fitbit app installed on my phone and I did notice a strong correlation of my BT connection improving once i deleted it, since then almost no dropped packets (due to BT connection).

Haven't noticed any weird scanning irregularities since either, although that wouldn't be effected by the fitbit app

old-square-eyes, 2018-02-11:
I have a few other BT parings, but not Fitbit. I'll remove them all as a precaution.

danpowell88, 2018-02-11:
Interestingly enough I didnt even have the fitbit paired (returned it to the store) but the app was somehow interrupting the BT stack it would seem

old-square-eyes, 2018-02-11:
I updated my other issue. I disconnected Putty and serial USB and got subsequent readings.

jstevensog, 2018-02-13:
I have had issues on my pixel 2 with pairings. Every power cycle, or BT
off/on makes the phone forget the pairing, even though xDrip indicates is
still valid. And sometimes other BT devices cause the pairing to break
somehow. Never had these issues with marshmallow.
Still, once I am permanently on xDrip+ I might be able to investigate more.
Cheers

old-square-eyes, 2018-02-14:
Hmmm. I seem to be experiencing the need to forget and rescan for the bridge on every bridge power cycle. Will do a few tests to verify.

jstevensog, 2018-02-14:
Does the BLE name go back to defaults? Are you using a hm-16/18?
Cheers