jstevensog/wixel-sdk #33: Cannot update transmitter id on 2.47e and 2.47f

kxwebster, 2017-12-17:
Running xDrip+ on the phone and xBridge wixel with an HM11 circuit as per Chris Bothello (CC2541) firmware v540. This has performed beautifully for a couple of years, but I cannot upgrade to 2.47e or 2.47f (See  diagnostics below.) and 2.47 seems to experience a lot of zombie disconnects.  Each instance started with a fresh install of Xdrip+ on the phone and a rewrite of the wixel and ended with a status request to confirm the stored settings. All done within minutes of each other, so no temperature differences. Also I did the checks referred to in issue "2.47e does not work" - both p1_6 and p1_7 are ok as expected because this xBridge functions OK at version 2.43, 2.46 and 2.47 Just the dropout rate at 2.46 and especially 2.47 appears to be more frequent. (or dropouts are more invisible/silent, difficult to manage due to improved sleeping).
Thanks for your work to date. Ken.
+++++++++++++++++++++++++++++++++++version 2.43 works OK++++++++++++
xBridge hardware circuit selected
batteryPercent val: 353
batteryPercent returning 0
No dex_tx_id.  Sending beacon.
OK+CONNOK+Get:06OK+Get:00O49359: sending beacon Now
Sending: ?K+Get:7D
Response: OK+Get:1EOK+Get:00OK+Get:25No dex_tx_id.  Sending beacon.
59423: sending beacon Now
Sending: ?
Response: ?ilooking for 6883233 (6J1W1)
waiting for packet on channel 0 for 0 
sProcessing Status Command
xBridge v2.43
dex_tx_id: 6883233 (6J1W1)
initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
battery_capacity: 0
current ms: 85247

+++++++++++++++++++++++++++++++++++version 2.46 works OK++++++++++++
AT+RESETxBridge hardware circuit selected
batteryPercent val: 351
batteryPercent returning 0
No dex_tx_id.  Sending beacon.
OK+CONNOK+Get:06OK+Get:00OK+Get35601: sending beacon Now
Sending: ?:7D
Response: OK+Get:1EOK+Get:00OK+Get:25No dex_tx_id.  Sending beacon.
45679: sending beacon Now
Sending: ?
Response: ?ilooking for 6883233 (6J1W1)
waiting for packet on channel 0 for 0 
sProcessing Status Command
xBridge v2.46
dex_tx_id: 6883233 (6J1W1)
initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
dex_tx_id_set: 1, got_packet: 0
battery_capacity: 0
current ms: 57187

+++++++++++++++++++++++++++++++++++version 2.47 works OK++++++++++++
Starting xBridge v2.47
Retrieving Settings
ATAT+NAMExBridgefc4fAT+RESETxBridge hardware circuit selected
batteryPercent val: 351
batteryPercent returning 0
No dex_tx_id.  Sending beacon.
OK+CONN30958: sending beacon Now
Sending: ?
Response: OK+Get:06OK+Get:00OK+Get:7DOK+Get:15OK+Get:00OK+Get:25No dex_tx_id.  Sending beacon.
40999: sending beacon Now
Sending: ?
Response: ?ilooking for 6883233 (6J1W1)
waiting for packet on channel 0 for 0 
sProcessing Status Command
xBridge v2.47
dex_tx_id: 6883233 (6J1W1)
initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
dex_tx_id_set: 1, got_packet: 0
battery_capacity: 0
current ms: 49884

+++++++++++++++++++++++++++++++++version 2.47e Fails to update TX-ID++++++++++++
Starting xBridge v2.47e
Retrieving Settings
Determining HM-1x baudrate 
trying 9600
11024 - send_data AT (2)
Sending: AT
Response: OK11065 - got OK
11065 - atBt() got OK
11066 - send_data AT (2)
11066 - send_data blocked on uart1 input
Sending: AT
Response: OK11087 - got OK
11087 - atBt() got OK
11088 - send_data AT (2)
11088 - send_data blocked on uart1 input
Sending: AT
Response: 11595 - atBt() Did not get an OK
12603 - send_data AT+NAMExBridgefc4f (18)
12604 - send_data blocked on uart1 input
?Sending: AT+NAMExBridgefc4f
Response: OK12641 - got OK
+Set:xBridgefc4f13131 - send_data AT+NOTI1 (8)
13131 - send_data blocked on uart1 input
Sending: AT+NOTI1
Response: OK+Set:113647 - send_data AT+RESET (8)
13647 - send_data blocked on uart1 input
Sending: AT+RESET
Response: OK+RESET?xBridge hardware circuit selected
batteryPercent val: 352
batteryPercent returning 0
No dex_tx_id.  Sending beacon.
OK+CONNble connected
25184: sending beacon Now
25184 - send_data ? (7)
25185 - send_data blocked on uart1 input
Sending: ?OK+Get:06OK+Get:00OK+Get:7DOK+GeNo dex_tx_id.  Sending beacon.
35697: sending beacon Now
35697 - send_data ? (7)
35697 - send_data blocked on uart1 input
Sending: ?No dex_tx_id.  Sending beacon.
46218: sending beacon Now
46218 - send_data ? (7)
46219 - send_data blocked on uart1 input
Sending: ?No dex_tx_id.  Sending beacon.
56739: sending beacon Now
56740 - send_data ? (7)
56740 - send_data blocked on uart1 input
Sending: ?sProcessing Status Command
xBridge v2.47e
dex_tx_id: 0 (00000)
initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
dex_tx_id_set: 0, got_packet: 0
battery_capacity: 0
current ms: 62166

No dex_tx_id.  Sending beacon.
67264: sending beacon Now
67264 - send_data ? (7)
67264 - send_data blocked on uart1 input
Sending: ?

+++++++++++++++++++++++++++++++++version 2.47f Fails to update TX-ID++++++++++++
Starting xBridge v2.47f
Retrieving Settings
Determining HM-1x baudrate 
trying 9600
11024 - send_data AT (2)
Sending: AT
Response: OK11065 - got OK
11065 - atBt() got OK
11066 - send_data AT (2)
11066 - send_data blocked on uart1 input
Sending: AT
Response: OK11087 - got OK
11087 - atBt() got OK
11088 - send_data AT (2)
11088 - send_data blocked on uart1 input
Sending: AT
Response: 11595 - atBt() Did not get an OK
12603 - send_data AT+NAMExBridgefc4f (18)
12604 - send_data blocked on uart1 input
?Sending: AT+NAMExBridgefc4f
Response: OK12641 - got OK
+Set:xBridgefc4f13131 - send_data AT+NOTI1 (8)
13131 - send_data blocked on uart1 input
Sending: AT+NOTI1
Response: OK+Set:113647 - send_data AT+RESET (8)
13647 - send_data blocked on uart1 input
Sending: AT+RESET
Response: OK+RESET?xBridge hardware circuit selected
batteryPercent val: 351
batteryPercent returning 0
No dex_tx_id.  Sending beacon.
OK+CONNble connected
23157: sending beacon Now
23157 - send_data ? (7)
23157 - send_data blocked on uart1 input
Sending: ?OK+Get:06OK+Get:00OK+Get:7DOK+GeNo dex_tx_id.  Sending beacon.
33669: sending beacon Now
33670 - send_data ? (7)
33670 - send_data blocked on uart1 input
Sending: ?No dex_tx_id.  Sending beacon.
44191: sending beacon Now
44191 - send_data ? (7)
44191 - send_data blocked on uart1 input
Sending: ?sProcessing Status Command
xBridge v2.47f
dex_tx_id: 0 (00000)
initialised: 0, sleep_ble: 1, dont_ignore_ble_state: 1, xBridge_hardware: 1, send_debug: 1, do_leds: 1
dex_tx_id_set: 0, got_packet: 0
battery_capacity: 0
current ms: 48912

No dex_tx_id.  Sending beacon.
54715: sending beacon Now
54715 - send_data ? (7)
54716 - send_data blocked on uart1 input
Sending: ?

tommy-gansta, 2017-12-17:
the same problem. have xbridge 2, 2.47e and kitty says no dex id

tepidjuice, 2018-08-29:
I have the same issue. Having this issue with 2.48b as well. Any fix?

jstevensog, 2018-08-29:
Hi All,
Strange, I just found this issue in my mail (don't get into git that often,
so rely on the emails).  So apologies to the earlier posters.
This is something I have never experienced (expect for the drop outs in the
earlier versions).  I have run each of these versions and never had an
issue setting the Transmitter ID.
I run both HM-10 and HM-16 modules in my tests, and always get a
transmitter ID, unless there is a bad solder joint corrupting the inbound
comms.  In one case, I had to replace the HM-1x module. as it was no longer
sending valid serial signals.
tepidjuice, can you please send me diagnostics and I will take a look.
3 separate bridges (including my reliable original production one) all
function fine on 2.48b.
Cheers