jstevensog/wixel-sdk #34: xBridge2.wxl invalid data in file error No.2

bordeb, 2018-01-22:
Hi! First, Jstevensog thank you on your great work!
I have the same issue today with latest 2.47e trying to write it to wixel.
When I try to open the .wxl app the Pololu Wixel Config Utility says exactly the same:
_Invalid data in file.
App File could not be opened: xBridge2.wxl_
I am using RPi - Raspbian (all latest updates) to do that. As a note: other .wxl files (for example jamorhams wixel-xDrip..) are opening without a problem.
I tried to download through cli using git , tried to download as zip file and than extract it on RPi, tried to do the same on Win and then transfer the file to RPi using Samba etc... Always the same error when I try to open wxl in the Wixel Utility.
Is there any solution for this?
I saw the closed issue #16 but can not reopen it so I am opening this as a new issue No.2. Apologies if this is not appropriate, please correct me if I am wrong.

bordeb, 2018-01-23:
After deeper analyzing the issue I have found that the problem is line endings of the .wxl file that are CR+CR+LF .
As such, the file can be opened without problem using Pololu Wixel Config Utility on Windows while those line endings are not acceptable by the same app Pololu Wixel Config Utility on RaspberryPi. The issue can be resolved (using for example Notepad++) through changing those CR+CR+LF line endings with regular CR+LF or just LF line endings - both those line endings fully works on both Windows as well as Linux (RPi) so my recomendfation would be to just use CR+LF or LF instead of CR+CR+LF if it is possible.

closed 2018-01-23 by bordeb

jstevensog, 2018-01-25:
Hello, and you are welcome.
I am not sure what the issue is with Raspian. It was similar for Apple
Mac's. It was a text file formatting error.
If you have dos2unix utility, try running that on the .wxl file first. I
run the file through sed as part of the make, which fixed the Mac issues.
Let me know if that helps.
Cheers

bordeb, 2018-01-28:
Hi @jstevensog and thank you very much for joining in even after I closed the issue. Appreciate this very very much! The issue is that in the latest version of the [wxl ](https://raw.githubusercontent.com/jstevensog/wixel-sdk/3491193680ed5b5198f7bef19e8f83195d21cca7/apps/xBridge2/xBridge2.wxl) file there seems to be CR+CR+LF line endings instead of regular CR+LF so that extra CR is producing the error on RPi while working on Win in the same time. I have easily fixed this with Notepad++ after figuring out what is the issue.. Maybe this extra unneeded CR can be removed somehow in the future builds what will consequently remove this issue on RPi.

old-square-eyes, 2018-01-28:
Just adding that I have this issue on the mac. I have to download the whole repo. Simply downloading the .wxl results in a flash fail. I'm guessing for the reasons described above.

jstevensog, 2018-01-30:
Thanks for the info, I'll check into it. May be something had gotten
corrupted in the make file that is meant to remove the CR and just leave
the LF on each line.
Cheers

bordeb, 2018-01-30:
Thanks man! Just a small hint, if you maybe find it useful: From what I can see, the present situation has one extra "unneeded" CR added before regular CR+LF ending. I think that one option is just to remove that first CR and leave CR+LF endings, or second option could be to remove both CRs in a row and just leave the LF endings. I think both options would produce good results, so it is a mater of choice which one is easier for the developer. Thanks once again for the great app!!

jstevensog, 2018-01-31:
Hi All,
I have found the issue, and uploaded a better Makefile and new xBridge2.wxl
for v2.47e.  I would ideally like to release v2.47f which has some changes
to the way frequency offsets are saved, and to my experience in production
is much more reliable.
Let me know if anyone has trouble with the new v2.47e file.  If you would
like to try v2.47f, it is in a branch with that name and also fixed.
Cheers

old-square-eyes, 2018-01-31:
Great thanks John. Have been on the lookout for a new firmware. Will try f now. Although I'm on a PC so can't verify the corruption fix for a bit.

bordeb, 2018-01-31:
Tnx @jstevensog everything fixed! Now is just a LF line ending which is great. I can confirm the fix is successful.
Also, thank you on provideing the f update. looking forward to test this.

old-square-eyes, 2018-02-11:
Just confirming. Right click and save as (windows 7) results in a corrupt file for me still. "Invalid data in file.
App File could not be opened: xBridge2.wxl"

jstevensog, 2018-02-13:
Ok, I'll look into it more. Not sure the right click save file accomplishes
what you think.
Cheers

jstevensog, 2018-02-13:
Tried loading my last commit xBridge2.wxl from my linux machine this
morning.  It handled it all fine.
But if you right click on the .wxl file on github and do a save as, you get
a file full of HTML.  This is probably the issue.
I doubt the wixel config utility on any platform would know to strip all
the HTML tags.

Probably the best option to get the file from the github directly would be
to copy and paste the text area of the .wxl display page into a text editor.
Don't know when this changed.  It used to work fine.
Cheers

old-square-eyes, 2018-02-13:
Thanks for following up John :) - It's not a major hindrance at all. It must be a Github thing. Weird though. You'd think you should be able to download a file untouched this way.

jstevensog, 2018-02-13:
indeed.