jstevensog/wixel-sdk #35: Issue with BT Module HM-10

niccoloserci, 2018-01-28:
Hi all,

after soldering the modules and uploading the 2.47 xBridge on the Wixel by the utility, I got no signals from the BT module (HM-10). 

What I get from Putty is what you can see on the 1st pic. If I try to upload the 2.0 xBridge version Putty becomes unresponsive. 

Any help on this? It's the second time I try to solder the kit and get the same results. On the specs the BT module seems to be fine.

Thanks
Samuele 
[image]

jstevensog, 2018-01-30:
Hello niccoloserci,
This is showing me three things.
1) The Phone has a BLE connection to the bridge.
2) The bridge does not yet have a TXID for the transmitter, and is sending
a BEACON packet to try and get it from the phone.
3) The phone is not responding to the BEACON packet.

There are two possible causes for this.
a) You have either the TXD and/or RXD lines between the Wixel and the HM-10
wrongly connected.
b) The HM-10 module is a non-genuine JNHuaMao module and is not responding
correctly to an AT command being sent to it to determine it's baud rate
setting.

xBridge2 when it first starts up sends AT at different baud rates until it
gets an OK response from the module.  The default is 9600 baud if it gets
no response, which is also the default for genuine JNHuaMao HM-10 modules.

If you can please use PuTTY to capture the initial boot up log of the
xBridge2, it will show clearly whether or not the wixel is able to
determine the baud rate or not.  Or you can load the usb-serial.wxl file on
the wixel and use PuTTY to try and send AT to the HM-10 to see if you get
an OK response.
You will need to try different baud rates to make sure you get the response.

Please carefully check your wiring to the diagram in the xBridge2.pdf file,
as incorrect wiring/soldering is the most common cause of this issue.
Cheers

niccoloserci, 2018-01-30:
Hi John,
thanks a lot for your reply and clarifications. Also, thanks for your efforts for the community, xDrip/xBridge has really changed my life! I'll try to check what you suggested in the next few day.
Cheers!
Samuele

jstevensog, 2018-01-30:
Hi Samuele,
You are most welcome.  They have both change my life as well ;).

My apologies to all for not having made progress of late.  A lot going on
in the family and personally has meant little time for this.
I do have two things I wish to achieve by the end of February.
1) Packet Cache as per savek-cc, but done more in line with the original
intent of the xBridge protocol standard.
2) Channel Scanning.  Yes it has been a long road, but Denis Lamarche has
sent me a file that seems to have cracked it.  I want to stitch it into the
xBridge2 code and really test it out.  It will be an awesome addition.

Cheers

danpowell88, 2018-02-03:
Appreciate the work John, 

You probably told me once before but what does the channel scanning stuff improve over how it works already?

Cheers

jstevensog, 2018-02-04:
Better packet capture is the major difference. While I have it pretty good
right now, it could always be better.
Cheers