jstevensog/wixel-sdk #36: One xBridge2 as data source for two (multiple) xDrip+ phones (devices)

bordeb, 2018-01-30:
My sincere apologies if this is not the best and most logical question, but I would really like to know from the experienced ones if there is some proper (feasible, secure) good way of getting two or more phones with xDrip+ app to connect to the same xBridge2 as their collector. I noticed that after the data is transferred between xBridge2 and xDrip, the xBridge2 goes to sleep. I am asking this from the perspective of redundancy in the possible practical situation when one has two phones, or maybe phone and cheap Chinese Android (but NOT Wear) smartwatch or...? The question is not targeted to Master + Follower setup. Also, I understand that in such a situations as described above the one should always calibrate both xDrip+ phones in order to have comparable and consistent devices/results.

jstevensog, 2018-01-30:
Hi Bordeb,
This is actually not possible because of BLE.  Each phone must connect to
the bridge using BLE.  First phone to connect will ensure the other cannot
connect.  If the bridge disconnects from the first phone, it is a race as
to which phone will reconnect first, and if the second phone reconnects
first, the HM-1 module will lose that first phone's connection.  This a BLE
limitation, and a race condition between the phones.
Whilst it might be possible using some intercommunication between the
phones, it would be a pretty fancy piece of coding, and I doubt would be
reliable.

One other possibility is to recode xBridge to use two HM-1x modules on
different serial IO ports on the wixel.  The code would need to try to get
an ACK from both ports before going to sleep.  That would be doable.

However, I am barely getting the time to do more important things with the
bridge code right now, and cannot spend time trying this out.  Happy if
someone else wants to give it a go.
Cheers

bordeb, 2018-01-30:
Hi!! One more time, thank you @jstevensog very much on your fast, precise, thorough and very descriptive answer. You sad it all!
Unfortunately my coding/programming knowledge for now is not even near the level necessary for trying such things as you described. :-) Now I understand completely and am going to close this issue. tnx!!

closed 2018-01-30 by bordeb