jstevensog/wixel-sdk #38: Wixel-HM10 as a G4 to G5 bridge ?

gszakacs, 2018-02-22:
How difficult would it be to convert the xBridge into a transceiver that receives G4 data (simpliciTI) on the wixel side and communicates as a G5 (BTLE) on the HM-10 side ?

gszakacs, 2018-02-23:
The battery replaced G5 only works with xDrip+ but it does not integrate with pumps for predictive auto-suspend or other hybrid control features after the battery life is over 112 days. G5 needs to be purchased 4 times a year vs a rechargeable G4 that runs for years.

jstevensog, 2018-02-24:
Yeah. I definitely hear you on that point. G5 was definitely a bad idea
without making it rechargeable. Hm-1x wouldn't be the best choice for this.
Would be worth thinking about a different SoC that is much more
configurable.
Cheers