jstevensog/wixel-sdk #39: Never sleep xbridge

Antonio2001, 2018-03-15:
Good morning from Spain. My xbridge works fine for a while and suddenly starts to lose readings and there is no way to make it work (I have to "forget device", reboot xbridge, and restart all the "bluetooth scan" process again and, If I'm lucky, it works again for a while)
I've been reading several old post and they talk about a "never sleep" mode to prevent the xbridge from sleeping and this way keep the bluetooth up and running (I tried with B option in Wixel command line but it doesn't seem to work. I don't know if the bluetooth module is a jnhuamao oringinal but I updated it to the last firmware version from its page, HM10-v603). Is it possible to post a xbridge2.wml that is always on (as the xdrip)? Can I install xdrip firmware in xbridge?
Thank you so much for your help
Kind regards

jstevensog, 2018-03-15:
Hola Antonio,
I visited your beautiful country last year with my wife and daughter.
There is not a "never sleep" mode, but there is a setting which will NOT
turn off the HM-1x bluetooth device, which is what the "b" command does.
However, I have found issues with Android GATT, depending on phone model
and Android versions, that does this.  I have a Pixel 2 these days and it
sometimes gets into a problem where it does not accept the data packet from
xBridge2.  The two most common causes of this are updates to Google apps
and Android components, and turning BT off and on or power cycle of the
phone.  With BT on and off and phone restart a forget and rescan is the
only way to get it working again.  Last night, after some updates, it
stopped receiving packets for an hour in the middle of the night, and then
started working again.  Usually, once it stops, it stops for good (unless I
am just too impatient).

Can you tell me if when you have hit the 'b' command, does the bridge say
sleep_ble off?  And if it does, does the BLE connection to the phone stay
on continually?
I would not like to hack a special version of xBridge if the 'b" command is
working as it should.
I will try to test this out on one of my other bridges, but it won't be for
a couple of days.  I have tried very hard to program around this Android
BLE issue over the years, but have not been able to nail it completely.  I
am still running the old xDrip rather than xDrip+, but when I get a chance
to work on running it on my Dev phone, I will see if there is an issue on
the xDrip+ side.  I friend in NZ had a similar issue with xDrip+ after a
recent nightly build, but before that it was working well.

Cheers

Antonio2001, 2018-03-16:
Hello Jonh. Thank you for your help and quick reply and for sharing your work with us. I'm happy you enjoyed your trip to Spain, quite a long travel from Australia...
Yes, I tried B command and the xbridge is always connected but keeps on loosing data. I have problems even connecting for the first time, waiting for the first three readiness, xbridge misses readings quite often. I installed xdrip firmware and it doesn't work... I'm using an xdrip now and probably will remove the resistors to convert the xbridge in a xdrip. 
I have found a xbridge3.wxl from 2015, I think. Shall I give it a try?
Thank you again
Kind regards

jstevensog, 2018-03-17:
Hi Antonio,
Do not use xBridge3.  It is still a work in progress.
If you installed xdrip firmware on the xBridge2 circuit, it will not work.
The difference is in the wiring of the HM-1x module.  In xdrip it is always
powered on.  In xBridge, it is powered on only when the wixel is awake,
unless the B command is issued on the console.

Is your xdrip you are using a different wixel entirely to your xBridge?
The two things that this can be are:
1) Your phone and Android version don't deal well with the BLE
disconnecting.  This is not likely if you are using an xDrip and it is
working OK.  However, you can test this out by moving your xDrip out or
range of your phone, waiting 10 minutes, and moving back in range.  If the
connection establishes and works from then on, the issue is probably not
the phone.

2) The xBridge wixel is faulty.  We had a bad batch a while ago (a year or
more?) in Australia that simply would not work reliably, as the crystals on
them were not stable.  The supplier replaced all the faulty ones.

If you have a reliable xDrip circuit, you could try converting it to an
xBridge circuit without the resistors (just move the +ve wire to the HM-1x
to the correct wixel pin for xBridge) and see if that works.  There are far
more advantages in xBridge2 (and soon to be xBridge3) than the xDrip
circuit.

Cheers

Antonio2001, 2018-03-17:
Thank you again, John. Yes, I have a working xDrip and a probably faulty xBridge. I'll try to make it work or turn the xDrip in a xBridge.
Regards

Antonio2001, 2018-03-23:
Hello. I found an old post (yes, again...):
https://github.com/jstevensog/wixel-sdk/issues/27
A gay with the same problem said xbridge2.wml version 2.47d worked fine for him but 2.47e didn't. You posted 2.47f and asked him to try it (this was the last post). I've tried 2.47d and it works, 2.47f (or 2.47e, the one in the master branch) and it doesn't, looses all the packages and has problems to connect. In the post you said 2.47e should work the same as 2.47d because it has only a few improves, but at least for me it doesn't.
Just to let you know, I hope it can help.
Thank you so much for your help, your work and for sharing it with us.
Kind regards

Antonio2001, 2018-07-23:
Good morning again. It seems there is a new (or at least new to me) terminal command. Pressing B will toggle sleeping bluetooth on/off. When I do it bluetooth's led keeps on and the xbridge is always connected but once the xbridge wakes up and sleeps again the bluetooth turns off (terminal says it's on but led is off and xdrip says "Disconnected"). How can I kep it on to avoid packet loss???
Thank you and regards

jstevensog, 2018-07-24:
Hi,
I was not aware that there was an issue with the B command.  I will
investigate.  Which version of xBridge have you seen this issue with?
Cheers