jstevensog/wixel-sdk #40: Simplified circuit

Antonio2001, 2018-09-11:
Good morning. Probably it has nothing to do with the new circuit (as far as I know the only modification is the connection between P1_2 an State). In xbridge 2.33 to 2.47d I have no problems with sleeping and wake up, but the wixel takes two minutes to sleep after sending a package to the phone: "waiting for ack" From 2.47e to 2.48b xbridge loses packages and doesn't wake uo as expected, but it always sleeps after sending a package and receiving the ack from the phone. Could it has something to do with the new circuit?
As always, thank you so much for your work and help
Regards

Antonio2001, 2018-09-12:
Good morning jstevensog. I'm afraid I can't send you the output, PuTTy stops working once the xbridge sleeps, the terminal freezes. I have read it can be due to wixel drivers or faulty wixel. I have made all the comprobations you said in another post with BL Scanner and the bluettoth module seems to be genuine, maybe the wixel is damaged :-( I'm not good at soldering, I have two bridges (one with the classic circuit, always on) and I had problems to find someone who could solder them for me, but I'll buy a new wixel and try to replace it (get replaced, I mean)
Thank you so much for your help

jstevensog, 2018-09-15:
Hi Antonio2001,
You are very welcome.  There were a batch of wixels some time ago that had
crystal stability issues.  And soldering well is key to a reliable circuit.
If you are doing it yourself, make sure you heat the wire and pad before
you apply solder.  Do not apply the solder to the tip of the soldering
iron.  Always let the wire and pad melt the solder and flux, and let it run
onto the parts.
It takes a little practice, but it is easy once you get the hang of it.
Also, make sure you "tin" the wire before you solder it to the pad.  This
means to heat the wire with the soldering iron, and apply just a small bit
of solder to the wire and let it melt into the wire.  This helps make a
very good and reliable joint.
Hope this helps out, and you have success.
Cheers