jstevensog/wixel-sdk #41: Transmitter serial to source/target radio address

Mitya1983, 2018-11-21:
Hi John,

I'm trying to make same bridge for my wife but using Arduino platform with C++ (I'm junior developer and this is for my self development mainly, and Arduino is more available in Ukraine) and I'm wondering how do you receive source address in ?? ?? ?? ?? format to read the data? If this is not a secret of course.

Thanks for your time.

Mitya1983, 2018-11-22:
Hi John,

I was hoping to catch the data from transmitter using Arduino. After that Arduino should decode the data to binary values (glucose rates) and depending on setting transmit those data to any compatible device via Bluetooth, WiFi, GSM if needed. 
Had searched through web and it seems that you are right about Arduino and SimpliciTI - they are not "friends"))) 
I found that Wixel has some extension interface for Arduino platform but haven't studied it yet.
Anyway if have to buy Wixel I think that I will by one. 

Arduino is more familiar to me.

Thank you one more time.

jstevensog, 2018-11-22:
Hi Dymtro,
Sorry, I am assuming that is your first name, I really should have asked.
You may be better off looking at a different device that is only a
programmable SimpliciTI radio.  I forget the device number.  This you could
easily connect to the Arduino, program the radio channels through the
addressing scheme and capture packets that way.  I will try and figure out
the chip number.  There may already be Arduino shields with this chip on
it.  IT would then be very easy to replicate what I have done on the wixel
to capture packets.

I would say that it is likely the wixel is a better option, as it's sleep
modes save a lot of battery drain.  I am not as familiar with this aspect
of the Arduino.

Cheers

closed 2022-09-16 by Mitya1983