jstevensog/wixel-sdk #42: Android 9 and xBridge compatibility

JonnyJohnson, 2019-04-27:
Have not found anything useable about this: I heard and read about issues when Android 8 was released in combination with the xBridge (packet and connection losses). Is there a similar issue with Android 9?

old-square-eyes, 2019-04-28:
I recently fired up my bridge and had no issue on my S8, Pie. I only tested it on a dev rig for about half a day though.

jstevensog, 2019-04-28:
I have been running 9 since it's release to my Pixel 2 production phone
with no real issues with connection and missed packets.
On occasion, I do have to restart the collector on xDrip, but that is
nothing new on any Android phone.  9 has been more stable than anything,
though I would say 8 was more so.
Can you please connect your bridge to a serial terminal and take a capture
of the debug log so I can review?
I may be able to point you in the right direction with more information.
Cheers.

jstevensog, 2019-05-01:
Hi JonnyJohnson,
Do you still have issues with this?  Please let me know.
Cheers

JonnyJohnson, 2019-05-02:
I am afraid of such issues (had problems with an older phone after updating to Android 8), so that I have not installed the update since the notification came up. My new phone seems to have better Bluetooth connectivity (no packet loss with v2.48b anymore), but I don't want to risk anything :)

jstevensog, 2019-05-02:
I have not had any issues using xBridge2 and any Android updates as far as
BLE connectivity is concerned.
Android and BLE have had a long and troubled history.  But, I usually find
every Android update improves things rather than breaks them.
Especially BLE.  Bluetooth itself has issues with upgrades, and this is why
you would have read about the G5/G6 issues with Android upgrades.
But I do understand where you are coming from.
I run an older code branch of xDrip+ because I am working on improvements
to the Pebble watch faces, and frankly don't need a lot of the new
features.

Very recently I am having packet loss issues, and am about to refresh my
xDrip+ repo and rebuild.
What is happening is that xDrip+ is acknowledging the packet from xBridge,
but not processing it for some reason.  Restarting the collector in the
System Status page always fixes it.
Since I have not changed my apk of xDrip+ for some time, it is likely that
an update has broken a hook or introduced some other condition that my
current apk runs into.
Anyway, a rebuild will prove it.  If that doesn't fix it I will investigate
deeper.

I have been running various versions of xBridge2 for over 5 years.  In that
time, I have had no update to Android break anything with it.  Each version
has only been improvements to the bridge packet capture, caching, and
battery life.
I am still running my original wixel and HM-1x, and have spares, one with a
HM-1x and one with a HM-17, for development and testing.  Never had any
issues with BLE connections with any of them, with any version of Android.
I have had issues with the xDrip code bases, and with packet capture on the
bridge which are mostly resolved these days.
Just have to refresh my xDrip+ apk every so often to get new features and
make sure it is built against the latest Android SDK.
Cheers

JonnyJohnson, 2019-05-03:
Sounds great. I will give it a try as you already have had good experiences with Android 9 and xBridge2.
Thanks!

closed 2019-05-03 by JonnyJohnson

jstevensog, 2019-05-05:
You're welcome.
Also, I updated rebuilt my xDrip+ on Friday, and the issues with xDrip+
acknowledging the packet, but not processing it, have now resolved.
Cheers