jstevensog/wixel-sdk #43: Android 10 and xBridge compatibility

JonnyJohnson, 2019-12-30:
Have not found anything useable about this: I heard and read about issues when Android 8 was released in combination with the xBridge (packet and connection losses). Is there a similar issue with Android 10 as it has been working perfectly with Android 9?

jstevensog, 2020-01-05:
Have used Android 10 with xbridge for over a year with no issues.  I have
recently moved to a G5 which has more issues, and may go back.
The issue with Android 8 was in making sure Location Services was enabled
otherwise the BLE would not join for some reason.  The same is required for
Android 10.
Cheers

jstevensog, 2020-01-05:
One other thing.  Make sure you are using the real JNHuaMao HM-1x modules
in your xBridge and ensure you have the most up to date firmware installed
on them.
We have had issues with non genuine HM-1x and older firmware version.  The
xBridge doco will tell you how to check this.
Cheers

JonnyJohnson, 2020-01-07:
Perfect, thanks. 
Now I safely can run the update.

closed 2020-01-07 by JonnyJohnson